RetroPie/RetroPie-Setup #2682: lr-pcsx-rearmed fails to build on XU4

tommitytom, 2019-04-19:
Firstly, I apologise for posting this here rather than on the forums, but I can't post on there because the confirmation email is never sent, and I am unable to verify my email address.

I used the latest setup script on a fresh install of Ubuntu MATE 18.04.1 minimal, on an odroid XU4.  Everything compiled fine except for lr-pcsx-rearmed.

Prep log:
```
RetroPie-Setup version: 4.4.12 (d35e619)
System: Linux odroid 4.14.111-158 #1 SMP PREEMPT Tue Apr 16 12:26:31 -03 2019 armv7l armv7l armv7l GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'lr-pcsx-rearmed' : Playstation emulator - PCSX (arm optimised) port for libretro
= = = = = = = = = = = = = = = = = = = = =

~/RetroPie-Setup/tmp/build/lr-pcsx-rearmed ~/RetroPie-Setup

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-pcsx-rearmed' : Playstation emulator - PCSX (arm optimised) port for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --recursive --depth 1 "https://github.com/libretro/pcsx_rearmed.git" "/home/retropie/RetroPie-Setup/tmp/build/lr-pcsx-rearmed"
Cloning into '/home/retropie/RetroPie-Setup/tmp/build/lr-pcsx-rearmed'...
HEAD is now in branch 'master' at commit 'e1d8eb0f131f9e7e838ec5658345e51ebcc3179c'
~/RetroPie-Setup
~/RetroPie-Setup/tmp/build/lr-pcsx-rearmed ~/RetroPie-Setup
```

The following error is thrown:

```
libpcsxcore/new_dynarec/backends/psx/pcsxmem.c:18:2: error: #error the dynarec is incompatible with Thumb functions,
 #error the dynarec is incompatible with Thumb functions,
  ^~~~~
libpcsxcore/new_dynarec/backends/psx/pcsxmem.c:19:2: error: #error please add -marm to compile flags
 #error please add -marm to compile flags
  ^~~~~
Makefile:279: recipe for target 'libpcsxcore/new_dynarec/backends/psx/pcsxmem.o' failed
```
I can post the full log if needed.

AbiuPlayer, 2019-04-19:
Need to add -marm to __default_cflags on line 334 of system.sh

tommitytom, 2019-04-19:
That did the trick, cheers!

hhromic, 2019-04-20:
@tommitytom  we just fixed this upstream, so you shouldn't need to add `-marm` to the default cflags anymore. Would you be able to revert this change and try to build from source to confirm? Thanks.
If it doesn't work, please provide us with any error you get.

tommitytom, 2019-04-21:
Tested, all good! :)

closed 2019-04-21 by joolswills