RetroPie/RetroPie-Setup #2688: FBAlpha  has changed to FBAlphaFB

LodanZark, 2019-05-02:
The fbalpha fork has changed to fbalphafb, probably due the blacklash related with licensing issues in the capcom home arcade case: [https://www.thegamer.com/capcom-home-arcade-emulator-open-source-backlash/
](https://www.thegamer.com/capcom-home-arcade-emulator-open-source-backlash/
)

lr-fbalphafb: https://github.com/libretro/FBAlphaFB

dankcushions, 2019-05-02:
this is an on-going discussion: http://neo-source.com/index.php?topic=3342.135

i wouldn't change anything yet

LodanZark, 2019-05-04:
Maybe the FBAlphaFB could be added to the as an experimental core or something.
I suspect that the older repo  it will not get more updates anytime soon  specially since the most active maintainers (barbudreadmon and dink) already moved to the new  repo.

barbudreadmon, 2019-05-05:
It's a bit too soon to change anything : 
- FBAlphaFB is just a temporary name
- the new repo is just a placeholder, it doesn't have a working libretro port yet (i've been waiting for the definitive new name to finish that stuff)

As usual, i'll let you know when anything major happens.

LodanZark, 2019-05-14:
Closed: https://github.com/barbudreadmon/RetroPie-Setup/commit/fd7af8047a07ea8cf34a26e3fcf658ab76c05d1f

closed 2019-05-14 by LodanZark