RetroPie/RetroPie-Setup #2700: Recent (compled from source) reicast crashes on start

romanbsd, 2019-05-12:
The crash occurs (at least on rpi3) with the following info:

`SIGSEGV @ 0 -> (nil) was not in vram, dynacode:0`

This patch fixed it for me:

```diff
diff --git a/core/rend/gles/gles.cpp b/core/rend/gles/gles.cpp
index 6965ec97d..8aab28fe0 100644
--- a/core/rend/gles/gles.cpp
+++ b/core/rend/gles/gles.cpp
@@ -560,7 +560,10 @@ GLuint fogTextureId;
 					return false;
 				}
 #ifdef GLES
+// EGL only supports runtime loading with android? TDB
+#ifdef _ANDROID
 				load_gles_symbols();
+#endif
 #else
 				egl_makecurrent();
 				if (gl3wInit())
diff --git a/core/rend/gles/gles.h b/core/rend/gles/gles.h
index 2b907c156..4fb98e656 100755
--- a/core/rend/gles/gles.h
+++ b/core/rend/gles/gles.h
@@ -15,7 +15,9 @@
 #endif
 #include <GLES32/gl32.h>
 #include <GLES32/gl2ext.h>
+#ifdef _ANDROID
 #include "gl32funcs.h"
+#endif
 
 #ifndef GL_NV_draw_path
 //IMGTEC GLES emulation

```

More info here:
https://github.com/reicast/reicast-emulator/issues/1537

hhromic, 2019-05-12:
Shouldn't this be reported/fixed upstream at the reicast repo?

romanbsd, 2019-05-12:
No idea what the status of this (I know that it's not merged to master though, and the branch also has another commit which is not necessary). I posted it here, so at least people who want to make it work will have a workaround.

cmitu, 2019-05-12:
Running from the [fh/gles-dynload](https://github.com/reicast/reicast-emulator/tree/fh/gles-dynload) branch does indeed make the issue go away, I guess it's not committed to the master branch yet.

romanbsd, 2019-05-12:
This branch has another commit which is not needed, though.

joolswills, 2019-05-13:
This needs to be sent upstream - we don't maintain this code.

closed 2019-05-13 by joolswills

2play, 2019-05-15:
@romanbsd hello
Would you please send me the compiled binary with above fixes. I have tried but wont compile now.

romanbsd, 2019-05-15:
Here you go: https://transfernow.net/705w57e13ika

2play, 2019-05-16:
@romanbsd Thank you very much!

2play, 2019-05-17:
latest scriptmodule compiles ok!
tx!