RetroPie/RetroPie-Setup #2710: FBAlpha is now FBNeo

barbudreadmon, 2019-05-21:
The "FBAlpha (FB Edition)" name was temporary, the final name was finally decided and it's "FinalBurn Neo" (reference to both the term "new" and the dev community behind it at https://neo-source.com), FBNeo for short.

The repository was renamed to https://github.com/libretro/FBNeo.
The corename is now fbneo.
All internal stuff using the string "fba" has been changed to "fbneo", including core options and paths for bioses & other support files.

Sorry for the inconvenience.

joolswills, 2019-05-23:
I think I will need to rename the entire module then and rename existing configs. Thanks for the info. Will do this asap.

joolswills, 2019-05-23:
How does this look - https://github.com/RetroPie/RetroPie-Setup/pull/2712

barbudreadmon, 2019-05-24:
Seems fine, the only improvement i can see would be to rename entries in core options but i guess you don't have a function for that ?

closed 2019-05-28 by joolswills