RetroPie/RetroPie-Setup #2733: RetroPie-Setup wiki for lr-fbalpha / FinalBurn Alpha

jmccorm, 2019-06-09:
Hi, all.

I haven't been able to find lr-fbalpha (217 under the Main packages install/update/remove menu.  

As a new user to RetroPie, I'm sad to say that it took way too much research for me to finally determine that just two weeks ago you've renamed lr-fbalpha to lr-fbneo. I'll be honest, I didn't even _think_ that lr-fbneo might have been the full emulator because I assumed that the only reason to include a vendor's name in the title would be because itis a Neo-Geo specific version. (Of course, I am now fully informed.)

Anyhow, the request:
1. Could you please update the RetroPie-Setup wiki pages with the new package name? I know that it would be very appreciated by myself and others.
2. As a near-term solution only, would you consider putting a dummy package back into the "Manage main packages" menu? Perhaps it could explain that the package has a new name, and it could also allow them to remove a legacy lr-fbalpha package where applicable.

Thank you for listening,
A new RetroPie convert, now less confused. :)

dankcushions, 2019-06-09:
As per the text you deleted, please use the forums for such discussions.

But yeah, the wiki is user-editable. you could update the name yourself or wait for an experienced user to do it, but please be patient for the latter as this is a hobby for us :)

LodanZark, 2019-06-09:
Fba it's outdated and seems ditched by all main devs because the Capcom Home Arcade license situation, your better off move to Fbn and forgot fba exist...

closed 2019-06-09 by joolswills