RetroPie/RetroPie-Setup #2734: Video clips in my menus

Namp2005, 2019-06-09:
Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

This includes edit suggestions for the wiki. There are more people to help on the forum.

Once a problem has been verified on the forum, an issue can be opened here.

Please remove this text before posting.

Namp2005, 2019-06-09:
First time posting a problem lol, I'm new to all this stuff, but trying to make nice menus with video clips, have a 128 GB SD card.
After a while they just stop showing.it says I'm scraping them, is it because I'm using to much space?

cmitu, 2019-06-09:
As the issue template explains, please open a topic in the forum - the Github Issues is used for bug reports, not for user support.

closed 2019-06-09 by joolswills