RetroPie/RetroPie-Setup #2736: Updating prompts for github login

ejstembler, 2019-06-15:
I'm attempting to initiate an update while using an xbox controller and my tv as the display.  At some point in the update it stops and prompts for a github login:

```shell
= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'lr-glupen64' : N64 emu - GLupeN64 for libretro
= = = = = = = = = = = = = = = = = = = = =

git clone --depth 1 "https://github.com/loganmc10/GLupeN64.git" "/home/pi/RetroPie-Setup/tmp/build/lr-glupen64"
Cloning into '/home/pi/RetroPie-Setup/tmp/build/lr-glupen64'...
Username for 'https://github.com': 
```

cmitu, 2019-06-16:
You're using a very old version of RetroPie-Setup, the repository that you mention doesn't exist anymore and has been removed from the scriptmodule in [2017](https://github.com/RetroPie/RetroPie-Setup/commit/d9b6e902def9fddba5d93b3e7d05e8f990bbfa97).

joolswills, 2019-06-16:
Please use the forum for support as detailed when opening a ticket.

closed 2019-06-16 by joolswills