RetroPie/RetroPie-Setup #2738: Cannot build from source on vero4k due to mame4all

DBMandrake, 2019-06-16:
Attempting to install Retropie on an OSMC Vero4k and it seems that the installation from source fails due to lack of platform support in mame4all:

```
= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'mame4all' : MAME emulator MAME4All-Pi
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): libsdl1.2-dev rbp-userland-dev-osmc. I am trying to install them now.
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  libcaca-dev libdrm-amdgpu1 libdrm-dev libdrm-exynos1 libdrm-freedreno1
  libdrm-nouveau2 libdrm-omap1 libdrm-radeon1 libdrm-tegra0 libgl1-mesa-dev
  libglu1-mesa libglu1-mesa-dev libpthread-stubs0-dev libpulse-dev
  libpulse-mainloop-glib0 libslang2-dev libx11-dev libx11-xcb-dev libxau-dev
  libxcb-dri2-0-dev libxcb-dri3-dev libxcb-glx0-dev libxcb-present-dev
  libxcb-randr0-dev libxcb-render0-dev libxcb-shape0-dev libxcb-sync-dev
  libxcb-xfixes0-dev libxcb1-dev libxdamage-dev libxdmcp-dev libxext-dev
  libxfixes-dev libxshmfence-dev libxxf86vm-dev mesa-common-dev
  rbp-userland-osmc x11proto-damage-dev x11proto-dri2-dev x11proto-fixes-dev
  x11proto-gl-dev x11proto-input-dev x11proto-kb-dev x11proto-xext-dev
  x11proto-xf86vidmode-dev xtrans-dev
Suggested packages:
  libxcb-doc libxext-doc
Recommended packages:
  libx11-doc
The following NEW packages will be installed:
  libcaca-dev libdrm-amdgpu1 libdrm-dev libdrm-exynos1 libdrm-freedreno1
  libdrm-nouveau2 libdrm-omap1 libdrm-radeon1 libdrm-tegra0 libgl1-mesa-dev
  libglu1-mesa libglu1-mesa-dev libpthread-stubs0-dev libpulse-dev
  libpulse-mainloop-glib0 libsdl1.2-dev libslang2-dev libx11-dev
  libx11-xcb-dev libxau-dev libxcb-dri2-0-dev libxcb-dri3-dev libxcb-glx0-dev
  libxcb-present-dev libxcb-randr0-dev libxcb-render0-dev libxcb-shape0-dev
  libxcb-sync-dev libxcb-xfixes0-dev libxcb1-dev libxdamage-dev libxdmcp-dev
  libxext-dev libxfixes-dev libxshmfence-dev libxxf86vm-dev mesa-common-dev
  rbp-userland-dev-osmc rbp-userland-osmc x11proto-damage-dev
  x11proto-dri2-dev x11proto-fixes-dev x11proto-gl-dev x11proto-input-dev
  x11proto-kb-dev x11proto-xext-dev x11proto-xf86vidmode-dev xtrans-dev
0 upgraded, 48 newly installed, 0 to remove and 12 not upgraded.
```
Notice that it is trying to install rbp-userland-dev-osmc, which is a Raspberry Pi specific package on OSMC. For a vero4k it should be vero3-userland-dev-osmc. All (?) other emulator build scripts seem to have special casing for the vero4k which works, for example retroarch:

```
= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'retroarch' : RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): libudev-dev libxkbcommon-dev libsdl2-dev vero3-userland-dev-osmc libfreetype6-dev libavcodec-dev libavformat-dev libavdevice-dev. I am trying to install them now.

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'sdl2' : SDL (Simple DirectMedia Layer) v2.x
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): devscripts libudev-dev libibus-1.0-dev libdbus-1-dev fcitx-libs-dev vero3-userland-dev-osmc. I am trying to install them now.
Ign:2 http://ftp.debian.org/debian stretch InRelease
Hit:3 http://security.debian.org stretch/updates InRelease
Hit:4 http://ftp.debian.org/debian stretch-updates InRelease
Hit:5 http://ftp.debian.org/debian stretch Release
Hit:1 http://ftp.fau.de/osmc/osmc/apt stretch InRelease
Reading package lists...
Reading package lists...
Building dependency tree...
Reading state information...
The following additional packages will be installed:
  dbus fcitx-bin gir1.2-ibus-1.0 libaudio2 libdbus-1-3 libfcitx-config4
  libfcitx-core0 libfcitx-gclient0 libfcitx-qt0 libfcitx-utils0
  libfile-homedir-perl libfile-which-perl libgettextpo0 libglib2.0-bin
  libglib2.0-data libglib2.0-dev libibus-1.0-5 libmng1 libpcre16-3
  libpcre3-dev libpcre32-3 libqt4-dbus libqt4-xml libqtcore4 libqtdbus4
  libqtgui4 libxt6 qdbus qtchooser qtcore4-l10n
Suggested packages:
  default-dbus-session-bus | dbus-session-bus adequate autopkgtest
  bls-standalone bsd-mailx | mailx check-all-the-things cvs-buildpackage
  devscripts-el diffoscope disorderfs dose-extra duck faketime gnuplot
  how-can-i-help libauthen-sasl-perl libfile-desktopentry-perl
  libnet-smtps-perl libterm-size-perl libyaml-syck-perl mozilla-devscripts
  mutt piuparts ratt reprotest svn-buildpackage w3m fcitx nas libglib2.0-doc
  qt4-qtconfig
Recommended packages:
  at dctrl-tools dput | dupload fakeroot libdistro-info-perl
  libgit-wrapper-perl liblist-compare-perl licensecheck lintian patchutils
  python3-debian python3-magic strace wdiff debian-keyring equivs
  libsoap-lite-perl fcitx-data fcitx-modules qt-at-spi
The following NEW packages will be installed:
  devscripts fcitx-bin fcitx-libs-dev gir1.2-ibus-1.0 libaudio2 libdbus-1-dev
  libfcitx-config4 libfcitx-core0 libfcitx-gclient0 libfcitx-qt0
  libfcitx-utils0 libfile-homedir-perl libfile-which-perl libgettextpo0
  libglib2.0-bin libglib2.0-data libglib2.0-dev libibus-1.0-5 libibus-1.0-dev
  libmng1 libpcre16-3 libpcre3-dev libpcre32-3 libqt4-dbus libqt4-xml
  libqtcore4 libqtdbus4 libqtgui4 libudev-dev libxt6 qdbus qtchooser
  qtcore4-l10n vero3-userland-dev-osmc
```

looking at the depends function for retroarch it has specific recognition for the vero4k and mali:

```
function depends_retroarch() {
    local depends=(libudev-dev libxkbcommon-dev libsdl2-dev libasound2-dev libusb-1.0-0-dev)
    isPlatform "rpi" && depends+=(libraspberrypi-dev)
    isPlatform "mali" && depends+=(mali-fbdev)
    isPlatform "x11" && depends+=(libx11-xcb-dev libpulse-dev libvulkan-dev)
    isPlatform "vero4k" && depends+=(vero3-userland-dev-osmc zlib1g-dev libfreetype6-dev)

    if compareVersions "$__os_debian_ver" ge 9; then
        depends+=(libavcodec-dev libavformat-dev libavdevice-dev)
    fi

    # only install nvidia-cg-toolkit if it is available (as the non-free repo may not be enabled)
    if isPlatform "x86"; then
        if [[ -n "$(apt-cache search --names-only nvidia-cg-toolkit)" ]]; then
            depends+=(nvidia-cg-toolkit)
        fi
    fi

    getDepends "${depends[@]}"

    addUdevInputRules
}
```

However mame4all has no specific recognition for the vero4k which causes it to be assumed to be a raspberry Pi thus indirectly pulling in rbp-userland-dev-osmc:

```
function depends_mame4all() {
    getDepends libasound2-dev libsdl1.2-dev libraspberrypi-dev
}
```

It looks like function depends_mame4all() in mame4all.sh needs updating to recognise the vero4k and use the appropriate userland package ?

hissingshark, 2019-06-18:
Thanks for picking up on this.  As we've subsequently discussed on the OSMC forum:

Anything that tries by default to install RPi specific packages is potentially bad news (system breakage), to say nothing of the fact it isn't going to build or run.  Also anything reliant upon SDL 1.2 is a no show, as it has no Mali support.

As there is more interest on the platform now I'm going to submit a PR to exclude the more "volatile" modules on vero4k.

DBMandrake, 2019-06-21:
Looks good. Now I see that mame4all can't be built on vero4k blacklisting it seems the best thing to do.

closed 2019-06-21 by DBMandrake