RetroPie/RetroPie-Setup #2753: misleading controller configuration dialogue

thatso, 2019-06-30:
Please change the highly misleading controller configuration dialogue
> PRESS A BUTTON

This reads like "press any button" similar to "press any key" while in reality it means "press button A", not button B, C, D, X, Y or whatever.
It took my ages to figure out why the configuration was not working before I realized the real meaning of this misleading instruction.
Please change it to:
>  Press button **A**

and maybe additionally highlight the letter A or better make it in the form of a circle so there is no doubt the controller button A is meant.

closed 2019-07-01 by joolswills

thatso, 2019-07-01:
@joolswills : care to explain why you closed my issue without any comment?

joolswills, 2019-07-01:
Because you didn't follow the guidance when opening an issue

>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>This includes edit suggestions for the wiki. There are more people to help on the forum.

>Once a problem has been verified on the forum, an issue can be opened here.

thatso, 2019-07-01:
You surely have a strange way of showing your gratitude for people who try to bring obvious issues to your attention while at the same time defying the purpose of GitHub's bug tracker.
Please excuse my stupid attempt to improve the user experience, it will never happen again.
Enjoy the numerous new users to come who will be annoyed without end trying to figure out how to configure their controllers because you refused to clarify the corresponding dialogue.

joolswills, 2019-07-02:
We have these guidelines in place for good reason. No need for such an attitude. Feel free to bring it up on the forum.