RetroPie/RetroPie-Setup #2762: incorrect filename reference in RetroPie-Setup/scriptmodules/ports/dxx-rebirth.sh

c0d3h4x0r, 2019-07-06:
Updating dxx-descent from source always fails near the end because this script erroneously tries to move two files named `README.txt`, which do not exist.  The actual filenames produced by the build are `README.RPi`, not `README.txt`.

hhromic, 2019-07-06:
Why not directly provide a PR with the fix? :)

joolswills, 2019-07-07:
They removed the old readme files. I will adjust.

closed 2019-07-07 by joolswills

c0d3h4x0r, 2019-07-07:
> Why not directly provide a PR with the fix? :)

Because git is the most user-hostile VCS I’ve ever had the displeasure of using.

hhromic, 2019-07-07:
Sorry to hear that, for me is a complete opposite experience.
Anyway the issue should be fixed now, thanks for reporting!

c0d3h4x0r, 2019-07-07:
To be completely fair, it's more GitHub's particular way of using git that infuriates me.  They don't make it easy or obvious how to take changes you've locally committed a clone of a project and somehow convert them to a commit on branch on a fork of a project (which is apparently what's required to create properly-formed pull request).

Indeed, I'm glad it's fixed!

commit 5e1754d0 (2019-07-07): dxx-rebirth - README.txt files removed upstream so remove from install - fixes #2762