RetroPie/RetroPie-Setup #2764: sub-optimal binaries for Pi 3B+

c0d3h4x0r, 2019-07-06:
The Raspberry Pi 3B+ has an armv8 CPU, but in their infinite wisdom, the RPi Foundation have chosen not to rebuild Raspbian specifically for armv8 (presumably to maintain compatibility with older Pi models without having to maintain a forked set of binaries).  Therefore, `uname -m` incorrectly reports the CPU architecture as armv71 rather than armv8, which in turn causes `configure` scripts to wrongly configure for armv71 rather than armv8, which in turn causes any packages updated from source to only get optimized for armv71 rather than armv8.  The performance difference is noticeable on at least certain emulators and games.

I've worked around this for now on my Pi 3B+ by adding the following two lines (which include some other non-arm-specific optimization flags) to the bottom of `/etc/bash.bashrc` and rebooting:

`export CFLAGS='-O3 -march=armv8-a+crc -mtune=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard -ftree-vectorize -ftracer'
export CXXFLAGS=${CFLAGS}`

This causes all subsequent package compilations to produce better-optimized binaries for the Pi 3B+, yielding noticeably better performance for certain emulators and games.  For example, Samurai Shodown under retroarch was somewhat unplayable (occasional audio drop-outs and framerate drops) using the original binary package, and was totally unplayable (even worse audio drop-outs framerate drops) after updating from source without my changes, but it now runs fine (no audio drop-outs or framerate drops) after recompiling with my changes.

hhromic, 2019-07-06:
While it's true that currently the provided binaries are not optimised for RPI3, when building from source on any RPI3 device, RetroPie uses the following configuration:
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/system.sh#L315-L320

Which sets (among others) the flags:

    -march=armv8-a+crc -mtune=cortex-a53 -mfpu=neon-fp-armv8 -mfloat-abi=hard

This means that when building from source you do get optimised binaries. Are you sure this is not the case for your setup?  I do see the correct compiler flags being passed when building anything from source.

joolswills, 2019-07-07:
Also this should be on the forum first as detailed when opening an issue. Thanks.

closed 2019-07-07 by joolswills