RetroPie/RetroPie-Setup #2776: Add systems and auto configurations lr-flycast atomiswave,naomi

DOCK-PI3, 2019-07-13:
In the ficher plataforms.cfg add:

atomiswave_exts=".zip .lst .bin"
atomiswave_fullname="Samy Atomiswave"
atomiswave_platform="dreamcast"
atomiswave_theme="atomiswave"

naomi_exts=".zip .lst .bin"
naomi_fullname="Sega Naomi"
naomi_platform="dreamcast"
naomi_theme="naomi"

The core with the changes
[lr-flycast.zip](https://github.com/RetroPie/RetroPie-Setup/files/3388355/lr-flycast.zip)

note: the core only reads .lst .bin files, the .zip files do not reproduce them. Greetings from Las Palmas.

darksaviorx, 2019-11-14:
Flycast no longer recommends .lst and .bin. If you load a game in that format, it will warn you with the message:"Please upgrade to mame romsets or expect issues." 

The mame versions are in .zip and .zip+.chd formats. If making a separate system tab for it, I recommend for only the .zip extension to be visible.

joolswills, 2019-11-14:
@darksaviorx thanks. I will close this then as I don't want to add more systems for this - users can customise and add more systems if they want.

closed 2019-11-14 by joolswills