RetroPie/RetroPie-Setup #2792: SDL_VIDEO_RPI_SCALE_MODE=1 does not work with mupen64plus-gles2n64

dankcushions, 2019-07-24:
Paging @gizmo98 who has done some work on this plugin :)

I'd noticed that this plugin was always in display resolution (in my case, 1080p), unlike the other plugins that run in N64 resolution (320x240, or similar), which we then upscale to display resolution via our SDL hint `SDL_VIDEO_RPI_SCALE_MODE=1`

The reason this didn't work for gles2n64 is because this line is wrong: https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/emulators/mupen64plus.sh#L248
the plugin does not like spaces surrounding the equals sign, so was ignoring this setting, and using display resolution via https://github.com/ricrpi/mupen64plus-video-gles2n64/blob/master/src/OpenGL.cpp#L395

If you correct the script (to `iniSet "auto resolution=0"`), it correctly uses n64 resolution, but for some reason the SDL upscale does not work, and you're left with a tiny image in the bottom left corner of the screen.

Does the SDL upscale only work in an SDL context? I believe gles2n64 uses an OpenGL context (there is an SDL pathway in the code but we don't use it for whatever reason). It would be nice to get this using an upscaled low resolution image to be consistent with the other plugins (and perform a little better).

gizmo98, 2019-07-24:
All other plugins use windowed mode. Maybe i have linked rpi scaling and windowed mode.
https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/emulators/mupen64plus/mupen64plus.sh#L443

Try to replace M64VIDEO_FULLSCREEN:https://github.com/ricrpi/mupen64plus-video-gles2n64/blob/master/src/OpenGL.cpp#L412

dankcushions, 2019-07-25:
thanks @gizmo98, that gave me the hint I needed! I've implemented fullscreen on/off in gles2n64 here: https://github.com/ricrpi/mupen64plus-video-gles2n64/pull/28

once accepted, I will do the PR here to correct our iniSet

dankcushions, 2019-08-13:
fixed in https://github.com/ricrpi/mupen64plus-video-gles2n64/pull/28
and then https://github.com/RetroPie/RetroPie-Setup/pull/2803

thanks all!

closed 2019-08-13 by dankcushions