RetroPie/RetroPie-Setup #2800: Error instaling retroarch

hcisilio, 2019-08-02:
in retropie_setup.sh retroarch is not installed:

```
= = = = = = = = = = = = = = = = = = = = =
   40 Building 'retroarch' : RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators
   41 = = = = = = = = = = = = = = = = = = = = =
   42 
   43 Checking operating system ... Linux (Ubuntu 19.04 19.04)
   44 Checking for suitable working C compiler ... /usr/bin/gcc works
   45 Checking for suitable working C++ compiler ... /usr/bin/g++ works
   46 Checking for pkg-config ... /usr/bin/pkg-config
   47 Checking for availability of switch -std=gnu99 in /usr/bin/gcc ... yes
   48 Checking for availability of switch -Wno-unused-result in /usr/bin/gcc ... yes
   49 Checking for availability of switch -Wno-unused-variable in /usr/bin/gcc ... yes
   50 Checking function sd_get_machine_names in -lsystemd ... no
   51 Checking presence of package bcm_host ... no
   52 Checking function bcm_host_init in -lbcm_host ... no
   53 Checking presence of package egl ... 19.0.2
   54 Checking function ass_library_init in -lass ... no
   55 Checking function pthread_create in -lpthread ... yes
   56 Checking function pthread_key_create in -lpthread ... yes
   57 Checking function dlopen in -ldl ... yes
   58 Checking function socket in -lc ... yes
   59 Checking function getaddrinfo in -lc ... yes
   60 Checking existence of -lminiupnpc ... no
   61 Checking function fcntl in -lc ... yes
   62 Checking function getopt_long in -lc ... yes
   63 Checking presence of package alsa ... 1.1.8
   64 Checking presence of predefined macro AUDIO_SETINFO in sys/audioio.h ... no
   65 Checking presence of package rsound >= 1.1 ... no
   66 Checking presence of package libroar ... no
   67 Checking presence of package libpulse ... 12.2
   68 Checking presence of package sdl2 >= 2.0.0 ... 2.0.9
 69 Checking presence of package flac ... no
   70 Checking presence of header file mbedtls/entropy.h ... no
   71 Checking existence of -lmbedx509 ... no
   72 Checking existence of -lmbedcrypto ... no
   73 Checking presence of package libusb-1.0 >= 1.0.13 ... 1.0.22
   74 Checking presence of header file GL/gl.h ... yes
   75 Checking existence of -lGL ... no
   76 Forced to build with library -lGL, but cannot locate. Exiting ...
   77 rm -rf obj-unix
   78 rm -f retroarch
   79 rm -f *.d
   80 config.mk is outdated or non-existing. Run ./configure again.
   81 make: *** [Makefile:191: config.mk] Error 1
   82 ~/RetroPie-Setup
   83 Could not successfully build retroarch - RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators (/home/helber/RetroP      ie-Setup/tmp/build/retroarch/retroarch not found).
```

joolswills, 2019-08-03:
Sorry but you need to start a topic on the forum first as detailed when opening an issue. And please follow the guidance as we have no idea about your setup..thanks.

joolswills, 2019-08-03:
(Ubuntu 19.04 is not sufficient info).

closed 2019-08-03 by joolswills