RetroPie/RetroPie-Setup #2801: Error initializing SDL while using A/V Jack

COOLIRON2311, 2019-08-03:
Everything works fine when Pi is using HDMI output.
However, this error occurs when it's using the A/V jack output: 
```
lvl0:   Error initializing SDL!

lvl0:   Renderer failed to initialize!
lvl0:   Window failed to initialize!
```

Firmware: 6c3fe3f096a93de3b34252ad98cdccadeb534be2
Kernel: 4.19.57-v7+

joolswills, 2019-08-03:
Please use the forum as requested.

closed 2019-08-03 by joolswills

COOLIRON2311, 2019-08-03:
I did. 0 replies in 24 hours.

joolswills, 2019-08-03:
Well this isn't our job. Please be patient.

joolswills, 2019-08-03:
You likely didn't get a reply as you didn't provide any details of your setup as requested when registering on the forum. Eg retropie version. I have replied to your post.

Jackowski3k, 2022-01-24:
> 

whats the point of github issues then?