RetroPie/RetroPie-Setup #2820: Install script user selection

EmilyBjoerk, 2019-08-25:
On my HTPC I have a user that runs the HTPC frontend applications (ES, Kodi, games, etc). This user does deliberately not have root access as it auto logs in on the machine, if it did it'd allow anyone with physical access root and I don't like that (before you say, any one with physical access can just grab the drive and mount it in another machine, I'm using cryptfs so that's not a thing). Then I have a personal user that has root access that I can log in with to do administration.

When I launch the setup script for RP it requires sudo, so I cannot launch it under the htpc user. So I run it with sudo from my user but the install script then forcefully installs to my user instead of the htpc user.

Can we have an option in the setup to change the user to install for?

joolswills, 2019-08-25:
The user it installs as will need sudo access. Please use the forum as detailed when opening a ticket though.

closed 2019-08-25 by joolswills

EmilyBjoerk, 2019-08-25:
That's fine, I'll configure ES manually then but you are encouraging insecure setups.

However, I will not register a forum account for every piece of software I use. That is an unreasonable ask.