RetroPie/RetroPie-Setup #2822: OpenMSX RetroPie Setup - Script break emulator install

2play, 2019-08-27:
When i updated from binary **or** source on a pi3 it deletes all except the share folder in the
/opt/retropie/emulators/openmsx/
binary etc all gone /opt/retropie/emulators/openmsx/bin/openmsx

Same today! It seems the scriptmodule is messed up
https://user-images.githubusercontent.com/30235736/63760478-aab7ed00-c8bf-11e9-8b8e-09dcd5b92230.png

joolswills, 2019-08-27:
The openmsx binaries need rebuilding as they are faulty - I will do that (and will be able to confirm if the source install is ok or not)

joolswills, 2019-08-27:
I will also add a build check to the code, as it doesn't have one (which is why no error occurred when building the binaries).

joolswills, 2019-08-27:
I've already done it. Just testing.

joolswills, 2019-08-27:
It wouldn't be memory requirements since I build in a chroot with plenty of ram on the host system. I will sort this.

cmitu, 2019-08-27:
Can you please provide a source install log - post the log to pastebin.com.

2play, 2019-08-27:
tx for checking gents!

cmitu, 2019-08-27:
I tested the build when changing the module and it was fine, retested today on PI4 and PC and still works.The only thing is the memory requirements - if they're changed (higher) recently, source install might fail. 

I can amend the module with a binary check in the `_build` part.

joolswills, 2019-08-28:
The binaries should be sorted now if you want to retry that @2play. It may have failed building from source for @2play due to running out of memory (A log would be needed to confirm). I may force it to -j1 for the RPI as it struggles even with lots of additional swap. There was also a missing dependency for alsa. I will commit the module fixes later.

joolswills, 2019-08-28:
What would help would be a log (not an image), from building from source as requested earlier. logs are compressed and saved in `~/RetroPie-Setup/logs` and can be accessed over SSH etc.

joolswills, 2019-08-28:
I've not yet updated the script module.

2play, 2019-08-28:
@joolswills 
Thank you. I tested from binary and all good as it seems! on below setup/script

[image]

2play, 2019-08-28:
I used binary. 
i can run the source and send logs.  I deleted the img that had the failed one. But I can retest on it and revert as I have a backup

2play, 2019-08-28:
@cmitu 
i heard and read that but you seem not reading mine. 
yes it is from binary. If you read my txt above that source compiled and at the end did the same to the emulators folder. 
after cleaning all it left ONLY share folder again.
Im just stating the result before when i tried both.
I will test on the img that i was trying to update yesterday and happily post logs for your checking.
but maybe the new updated scriptmodule is ok and wont replicate my yesterday error.

reverting

2play, 2019-08-28:
ok will retry on the previous img. to be frank i tried also yesterday no latest clean retropie base 4.5.1, updated the script and installed from binary with same result.
installed today a new retropie setup to a clean raspbian and after your last message i tried from binary and went ok.
so this is my full feedback and actions taken so far.
retesting from source on my own img as promised and reverting with results @joolswills

cmitu, 2019-08-28:
> From Source before it seems once compiled to do the same steps as posted on my first post.

No, it's not - source install always compiles the package - the screenshot you posted is from a binary install. Please add a log from a source install, as @joolswills explained.

joolswills, 2019-08-28:
I reproduced an out of memory error on a rpi3 even with rpSwapOn 1500. 2000 seemed to be ok. Was very slow though due to the swapping.

2play, 2019-08-28:
hi @joolswills 
finally finished compiling after 2h 
[rps_2019-08-28_151605.log.gz](https://github.com/RetroPie/RetroPie-Setup/files/3552187/rps_2019-08-28_151605.log.gz)
on my img that had probs yesterday when i posted.
finished without failing BUT it left only a share folder in the emulators/openmsx folder
like yesterday

Clean boot, all default...
Here is the script

2play, 2019-08-28:
[image]

rather right after from binary

[image]

2play, 2019-08-28:
[rps_2019-08-28_174324.log.gz](https://github.com/RetroPie/RetroPie-Setup/files/3552217/rps_2019-08-28_174324.log.gz)
log from binary just afterwards

closed 2019-08-28 by joolswills

joolswills, 2019-08-28:
Was an out of memory issue. Should be sorted now but needs a lot of ram.

joolswills, 2019-08-28:
I didn't work out why the binaries were broken but I suspect it could have been due to just not having rebuilt them after @mitu fixed up openmsx previously.

commit 5ba8635f (2019-08-28): openmsx - fix build issues - fixes #2822
 * add libasound2-dev dependency
 * increase swap
 * add check to see build is successful