RetroPie/RetroPie-Setup #2825: unknown platform for Jetson nano

jetson-retropie, 2019-08-31:
When setting up retropie-setup in terminal. I'm getting this error for jetson nano.

Unknown platform - please manually set the __platform variable to one of the following: armv7-mali generic-x11 imx6 jetson-nano odroid-c1 odroid-c2 odroid-xu rpi1 rpi2 rpi3 tinker vero4k x86

joolswills, 2019-08-31:
Because it's not supported. Please use the forum first as detailed when opening an issue.

joolswills, 2019-08-31:
Actually I missed it the list, but this was added by a 3rd party. So it's untested - you will have to implement this yourself if you have a board that's not recognised. It's not an official platform we support though.

closed 2019-08-31 by joolswills

jetson-retropie, 2019-08-31:
Hi,
Thanks for informing, i will try out the board on my own and maybe with the 3party help. 

May I know would it be possible to gain support officially or this has to be 3rd party?

Regards
Joo Kuan

joolswills, 2019-08-31:
Just checked and we don't have jetson-nano support included at all yet. Did you patch / modify RetroPie-Setup - eg with this https://github.com/RetroPie/RetroPie-Setup/pull/2821 ?

```
cd RetroPie-Setup
sudo __platform=jetson-nano ./retropie_setup.sh
```

If that works and the emulators install and run please provide the output of `/sys/firmware/devicetree/base/model`

jetson-retropie, 2019-08-31:
Hey, It works!! thanks for the help

jetson-retropie, 2019-08-31:
here are the output for device tree

jetson@jetson-desktop:~$ echo $(cat /sys/firmware/devicetree/base/*)
cat: /sys/firmware/devicetree/base/aconnect@702c0000: Is a directory
cat: /sys/firmware/devicetree/base/actmon@6000c800: Is a directory
cat: /sys/firmware/devicetree/base/adma@702e2000bash: warning: command
substitution: ignored null byte in input
: Is a directory
cat: /sys/firmware/devicetree/base/adsp_audio: Is a directory
cat: /sys/firmware/devicetree/base/agic-controller: Is a directory
cat: /sys/firmware/devicetree/base/ahb@6000c000: Is a directory
cat: /sys/firmware/devicetree/base/ahub: Is a directory
cat: /sys/firmware/devicetree/base/aliases: Is a directory
cat: /sys/firmware/devicetree/base/apbmisc@70000800: Is a directory
cat: /sys/firmware/devicetree/base/arm-pmu: Is a directory
cat: /sys/firmware/devicetree/base/bluedroid_pm: Is a directory
cat: /sys/firmware/devicetree/base/bpmp: Is a directory
cat: /sys/firmware/devicetree/base/bthrot_cdev: Is a directory
cat: /sys/firmware/devicetree/base/bwmgr: Is a directory
cat: /sys/firmware/devicetree/base/camera-pcl: Is a directory
cat: /sys/firmware/devicetree/base/chosen: Is a directory
cat: /sys/firmware/devicetree/base/clock: Is a directory
cat: /sys/firmware/devicetree/base/clock@70110000: Is a directory
cat: /sys/firmware/devicetree/base/core_dvfs_cdev_cap: Is a directory
cat: /sys/firmware/devicetree/base/core_dvfs_cdev_floor: Is a directory
cat: /sys/firmware/devicetree/base/cpu_edp: Is a directory
cat: /sys/firmware/devicetree/base/cpufreq: Is a directory
cat: /sys/firmware/devicetree/base/cpuidle: Is a directory
cat: /sys/firmware/devicetree/base/cpus: Is a directory
cat: /sys/firmware/devicetree/base/dfll-cdev-cap: Is a directory
cat: /sys/firmware/devicetree/base/dfll-cdev-floor: Is a directory
cat: /sys/firmware/devicetree/base/dfll-max77621@70110000: Is a directory
cat: /sys/firmware/devicetree/base/dma@60020000: Is a directory
cat: /sys/firmware/devicetree/base/dma_test: Is a directory
cat: /sys/firmware/devicetree/base/dtv@7000c300: Is a directory
cat: /sys/firmware/devicetree/base/dummy-cool-dev: Is a directory
cat: /sys/firmware/devicetree/base/dvfs: Is a directory
cat: /sys/firmware/devicetree/base/dvfs_rails: Is a directory
cat: /sys/firmware/devicetree/base/eeprom-manager: Is a directory
cat: /sys/firmware/devicetree/base/efuse@7000f800: Is a directory
cat: /sys/firmware/devicetree/base/extcon: Is a directory
cat: /sys/firmware/devicetree/base/external-memory-controller@7001b000: Is
a directory
cat: /sys/firmware/devicetree/base/flow-controller@60007000: Is a directory
cat: /sys/firmware/devicetree/base/gpio@6000d000: Is a directory
cat: /sys/firmware/devicetree/base/gpio-keys: Is a directory
cat: /sys/firmware/devicetree/base/gpio-timed-keys: Is a directory
cat: /sys/firmware/devicetree/base/gps_wake: Is a directory
cat: /sys/firmware/devicetree/base/gpu: Is a directory
cat: /sys/firmware/devicetree/base/gpu-dvfs-rework: Is a directory
cat: /sys/firmware/devicetree/base/gpu_edp: Is a directory
cat: /sys/firmware/devicetree/base/hda@70030000: Is a directory
cat: /sys/firmware/devicetree/base/host1x: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000c000: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000c400: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000c500: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000c700: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000d000: Is a directory
cat: /sys/firmware/devicetree/base/i2c@7000d100: Is a directory
cat: /sys/firmware/devicetree/base/interrupt-controller: Is a directory
cat: /sys/firmware/devicetree/base/interrupt-controller@60004000: Is a
directory
cat: /sys/firmware/devicetree/base/iommu: Is a directory
cat: /sys/firmware/devicetree/base/kfuse@7000fc00: Is a directory
cat: /sys/firmware/devicetree/base/lens_imx219@RBPCV2: Is a directory
cat: /sys/firmware/devicetree/base/mailbox@70098000: Is a directory
cat: /sys/firmware/devicetree/base/max16984-cdp: Is a directory
cat: /sys/firmware/devicetree/base/mc: Is a directory
cat: /sys/firmware/devicetree/base/memory@80000000: Is a directory
cat: /sys/firmware/devicetree/base/memory-controller@70019000: Is a
[... 56 lines cut ...]
cat: /sys/firmware/devicetree/base/__symbols__: Is a directory
cat: /sys/firmware/devicetree/base/tegra-aotag: Is a directory
cat: /sys/firmware/devicetree/base/tegra-camera-platform: Is a directory
cat: /sys/firmware/devicetree/base/tegra-carveouts: Is a directory
cat: /sys/firmware/devicetree/base/tegra_cec: Is a directory
cat: /sys/firmware/devicetree/base/tegra_fiq_debugger: Is a directory
cat: /sys/firmware/devicetree/base/tegra-pmc-blink-pwm: Is a directory
cat: /sys/firmware/devicetree/base/tegra-supply-tests: Is a directory
cat: /sys/firmware/devicetree/base/tegra_udrm: Is a directory
cat: /sys/firmware/devicetree/base/tfesd: Is a directory
cat: /sys/firmware/devicetree/base/thermal-fan-est: Is a directory
cat: /sys/firmware/devicetree/base/thermal-zones: Is a directory
cat: /sys/firmware/devicetree/base/timer: Is a directory
cat: /sys/firmware/devicetree/base/timer@60005000: Is a directory
cat: /sys/firmware/devicetree/base/tlk: Is a directory
cat: /sys/firmware/devicetree/base/trusty: Is a directory
cat: /sys/firmware/devicetree/base/usb_cd: Is a directory
cat: /sys/firmware/devicetree/base/watchdog@60005100: Is a directory
cat: /sys/firmware/devicetree/base/wifi-antenna-tuning: Is a directory
cat: /sys/firmware/devicetree/base/xotg: Is a directory
cat: /sys/firmware/devicetree/base/xudc@700d0000: Is a directory
cat: /sys/firmware/devicetree/base/xusb@70090000: Is a directory
cat: /sys/firmware/devicetree/base/xusb_padctl@7009f000: Is a directory
nvidia,jetson-nanonvidia,tegra210 jetson-nano3448Mar 15
201909:52:07/dvs/git/dirty/git-master_linux/kernel/kernel-4.9/arch/arm64/boot/dts/../../../../../../hardware/nvidia/platform/t210/porg/kernel-dts/tegra210-p3448-0000-p3449-0000-a02.dts�B3448344804212190249020408004

mrcmunir, 2019-08-31:
@joolswills 
[image]
Returns NVIDIA Jetson Nano Developer Kit 

@jetson-retropie he said what name returned by cat /sys/firmware/devicetree/base/model .

jetson-retropie, 2019-09-01:
Hi mrcmunir,
with jetson-nano, ubuntu

i ran this command in terminal  ./retropie_setup.sh

jetson@jetson-desktop:~/RetroPie-Setup$ sudo ./retropie_setup.sh
Unknown platform - please manually set the __platform variable to one of
the following: armv7-mali generic-x11 imx6 jetson-nano odroid-c1 odroid-c2
odroid-xu rpi1 rpi2 rpi3 tinker vero4k x86

i try this
sudo __platform=jetson-nano ./retropie_setup.sh

and it works.

regards
Joo Kuan

mrcmunir, 2019-09-01:
@jetson-retropie In you system.sh has this line after *) ? 

[code]if grep -q "NVIDIA Jetson Nano Developer Kit" /sys/firmware/devicetree/base/model 2>/dev/null; then
                    __platform="jetson-nano" 

        else
              case $architecture in
                         i686|x86_64|amd64)
                        __platform="x86"
                        ;;
                   esac
                fi
              ;;
        esac
    fi
[/code]

Please see 
[code]https://raw.githubusercontent.com/RetroPie/RetroPie-Setup/ced3319b5d58a0910f8696feec836e85bd36edc8/scriptmodules/system.sh[/code]

I use Grep "name for detected _plattform=jetson-nano auto

jetson-retropie, 2019-09-02:
Hi Mrmuncir,
Sorry for late reply. Yes I have the line of code in system.sh

The issue is after installing retropie, I am unable to get into retropie setup via emulationstation.

To get in retropie. I have to amend system.sh 

Regards
Joo Kuan

mrcmunir, 2019-09-02:
@jetson-retropie  Well, it's strange, you just have to do sudo ./retropie_setup.sh

You should be able to install almost everything except those small cores that are not yet supported for aarch64 (i.e you needed mupen64 next core for aarch64) or small emulators that are armv7 i.e picodrive don't compiler  or some smalls packages for raspberry pi very specific .