RetroPie/RetroPie-Setup #2860: PowerBlock fails to build since installation updates

iasdeoupxe, 2019-10-03:
Haven't put a forum posting as there is no need to confirm this.

The changes in https://github.com/petrockblog/PowerBlock/commit/0f26e06826aa8d05131a1c313088ce65902bfc6d had removed the CMakeLists.txt causing the following build issue of the PowerBlock driver when updating RetroPie:

```
= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

Already on 'master'
Your branch is up-to-date with 'origin/master'.
Already up-to-date.
HEAD is now in branch 'master' at commit 'a7db2cb574a1fc046474bd1eca8d25288aaeab2c'
/home/pi

= = = = = = = = = = = = = = = = = = = = =
Building 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

CMake Error: The source directory "/opt/retropie/supplementary/powerblock" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
make: *** No targets specified and no makefile found.  Stop.
Could not successfully build powerblock - PowerBlock Driver (/opt/retropie/supplementary/powerblock/build/src/powerblock/powerblock not found).
```

Looks like https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/supplementary/powerblock.sh needs some rework for the new installation method mentioned in:

https://github.com/petrockblog/PowerBlock#installation

joolswills, 2019-10-04:
@petrockblog - you ok for me to look into this or would you prefer to ?

iasdeoupxe, 2019-10-04:
Sorry, forgot to mention that i also had to delete the "docs" folder in /opt/retropie/supplementary/powerblock` because it seems the installation had created / modified some files within the repository causing the `git pull` command to fail:

```
= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

/home/pi/RetroPie-Setup/tmp/build/powerblock /home/pi

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

Already on 'master'
M       doc/htmlDoc/html/doxygen.css
M       doc/htmlDoc/html/index.html
M       doc/htmlDoc/html/jquery.js
M       doc/htmlDoc/html/search/search.css
M       doc/htmlDoc/html/search/search.js
M       doc/htmlDoc/html/tabs.css
M       doc/htmlDoc/latex/doxygen.sty
M       doc/htmlDoc/latex/refman.tex
Your branch is up-to-date with 'origin/master'.
From https://github.com/petrockblog/PowerBlock
   30485df..a7db2cb  master        -> origin/master
 * [new tag]         release/1.5.0 -> release/1.5.0
 * [new tag]         release/2.0.0 -> release/2.0.0
 * [new tag]         release/1.6.0 -> release/1.6.0
 * [new tag]         release/1.6.1 -> release/1.6.1
 * [new tag]         release/1.6.2 -> release/1.6.2
 * [new tag]         release/1.6.3 -> release/1.6.3
error: Your local changes to the following files would be overwritten by merge:
        doc/htmlDoc/html/doxygen.css
        doc/htmlDoc/html/index.html
        doc/htmlDoc/html/jquery.js
        doc/htmlDoc/html/search/search.css
        doc/htmlDoc/html/search/search.js
        doc/htmlDoc/html/tabs.css
        doc/htmlDoc/latex/doxygen.sty
        doc/htmlDoc/latex/refman.tex
Please commit your changes or stash them before you merge.
Aborting
Updating 30485df..a7db2cb
HEAD is now in branch 'master' at commit '30485dfe9e0341a442ffdc1efa9048c2fcbfccd0'
/home/pi
Error running 'git pull' - returned 1
```

petrockblog, 2019-10-04:
@joolswills Thanks a lot for the offer. I have just created a pull-request (#2862 ) that should solve the issue. However, I have not tested those changes yet myself. Sorry! I should be able to test them during this weekend, though.

joolswills, 2019-10-04:
Cheers!

iasdeoupxe, 2019-10-11:
Confirming that #2862 / c8d27116ecf5c05c0abfe672b12a463bb4661af5 to fix the build issue:

```
= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

Already on 'master'
Your branch is up-to-date with 'origin/master'.
Already up-to-date.
HEAD is now in branch 'master' at commit 'a7db2cb574a1fc046474bd1eca8d25288aaeab2c'
/home/pi

= = = = = = = = = = = = = = = = = = = = =
Installing 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

Reading package lists...
Building dependency tree...
Reading state information...
git is already the newest version (1:2.11.0-3+deb9u4).
0 upgraded, 0 newly installed, 0 to remove and 7 not upgraded.
Cloning into 'PowerBlock'...
make: *** No rule to make target 'uninstallservice'.  Stop.
Starting powerblock (via systemctl): powerblock.service.
Installation of PowerBlock service done.
[SUCCESS] The PowerBlock driver binary is installed
[SUCCESS] The PowerBlock service is running
You can find the configuration file at /etc/powerblockconfig.cfg.
```

and

```
Oct 11 04:49:02 pi systemd[1]: Starting LSB: powerblock init script...
Oct 11 04:49:02 pi systemd[1]: Started LSB: powerblock init script.
Oct 11 04:49:02 pi powerblockservice: Starting PowerBlock driver, version 2.0.0
Oct 11 04:49:02 pi powerblockservice: Shutdown is ACTIVATED
Oct 11 04:49:02 pi powerblockservice: Shutdown Pin is 18
Oct 11 04:49:02 pi powerblockservice: Status Pin is 17
Oct 11 04:49:02 pi powerblockservice: Setting RPi status signal to HIGH
```

@petrockblog @joolswills I leave it up to you if https://github.com/RetroPie/RetroPie-Setup/issues/2860#issuecomment-538325954 worth a fix or if the issue should be closed as "workaround exists" (which is to make a ``git checkout --`` in /opt/retropie/supplementary/powerblock/).

petrockblog, 2019-10-12:
I fixed this with PR #2862. No workaround needed anymore.

closed 2019-10-12 by petrockblog

iasdeoupxe, 2019-10-13:
@petrockblog Please correct me if i'm wrong but i don't see something in #2862 which could prevent/fix the "Error running 'git pull' - returned 1" message shown in https://github.com/RetroPie/RetroPie-Setup/issues/2860#issuecomment-538325954

This seems to be caused by the old PowerBlock build script changing files within the /docs folder during build time where the git pull/checkout is failing afterwards.

joolswills, 2019-10-13:
If the powerblock module generates files that are committed to git we can workaround it. Will wait for petrockblog on this.

petrockblog, 2019-10-13:
I have just pushed a minor update that also takes care for possible local changes in the powerblock module directory (87b634271ad9a8f1c15157347a1306564aba4039).