RetroPie/RetroPie-Setup #2869: gitPullOrClone: git restore / reset local repo to avoid pull failures

iasdeoupxe, 2019-10-13:
Inspired by https://github.com/RetroPie/RetroPie-Setup/issues/2860#issuecomment-538325954 and the issue shown below gitPullOrClone could do a ``git checkout -- *`` (or similar) to avoid that a possible modification of local files (e.g. caused by the build scripts of the installed project) is causing an error during the installation.

```
= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

/home/pi/RetroPie-Setup/tmp/build/powerblock /home/pi

= = = = = = = = = = = = = = = = = = = = =
Getting sources for 'powerblock' : PowerBlock Driver
= = = = = = = = = = = = = = = = = = = = =

Already on 'master'
M       doc/htmlDoc/html/doxygen.css
M       doc/htmlDoc/html/index.html
M       doc/htmlDoc/html/jquery.js
M       doc/htmlDoc/html/search/search.css
M       doc/htmlDoc/html/search/search.js
M       doc/htmlDoc/html/tabs.css
M       doc/htmlDoc/latex/doxygen.sty
M       doc/htmlDoc/latex/refman.tex
Your branch is up-to-date with 'origin/master'.
From https://github.com/petrockblog/PowerBlock
   30485df..a7db2cb  master        -> origin/master
 * [new tag]         release/1.5.0 -> release/1.5.0
 * [new tag]         release/2.0.0 -> release/2.0.0
 * [new tag]         release/1.6.0 -> release/1.6.0
 * [new tag]         release/1.6.1 -> release/1.6.1
 * [new tag]         release/1.6.2 -> release/1.6.2
 * [new tag]         release/1.6.3 -> release/1.6.3
error: Your local changes to the following files would be overwritten by merge:
        doc/htmlDoc/html/doxygen.css
        doc/htmlDoc/html/index.html
        doc/htmlDoc/html/jquery.js
        doc/htmlDoc/html/search/search.css
        doc/htmlDoc/html/search/search.js
        doc/htmlDoc/html/tabs.css
        doc/htmlDoc/latex/doxygen.sty
        doc/htmlDoc/latex/refman.tex
Please commit your changes or stash them before you merge.
Aborting
Updating 30485df..a7db2cb
HEAD is now in branch 'master' at commit '30485dfe9e0341a442ffdc1efa9048c2fcbfccd0'
/home/pi
Error running 'git pull' - returned 1
```

joolswills, 2019-10-13:
This case should be fixed just for this module I think (can continue discussion on the other issue). I don't want a module wide change.

closed 2019-10-13 by joolswills

iasdeoupxe, 2019-10-13:
Haven't thought about this and it indeed makes sense. The building of a module probably should never touch files in a repository on its own and do that only in a not-checked in build directory, everything else sounds to me like a bug in the build scripts of the module itself.