RetroPie/RetroPie-Setup #2909: Issues building Mupen64plus through Retropie on Rpi4 during creation of "glsl_SpecialShadersFactory.cpp.o"

kylepd49, 2019-11-28:
Hello,
I used the fkms branch to install Retropie onto my Rpi4 I'm currently having issues finishing the build for Mupen64plus. It halts at 70% and presents the following information stating there is an issue during building: 

/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp: In member function ‘virtual void glsl::TexrectCopyShader::activate()’:
/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp:798:4: error: ‘glGetTexLevelParameteriv’ was not declared in this scope
    glGetTexLevelParameteriv(0, 0, GL_TEXTURE_WIDTH, &texWidth);
    ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp:798:4: note: suggested alternative: ‘glTextureParameteri’
    glGetTexLevelParameteriv(0, 0, GL_TEXTURE_WIDTH, &texWidth);
    ^~~~~~~~~~~~~~~~~~~~~~~~
    glTextureParameteri
/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp: In member function ‘virtual void glsl::TexrectColorAndDepthCopyShader::activate()’:
/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp:836:4: error: ‘glGetTexLevelParameteriv’ was not declared in this scope
    glGetTexLevelParameteriv(0, 0, GL_TEXTURE_WIDTH, &texWidth);
    ^~~~~~~~~~~~~~~~~~~~~~~~
/home/pi/RetroPie-Setup/tmp/build/mupen64plus/GLideN64/src/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp:836:4: note: suggested alternative: ‘glTextureParameteri’
    glGetTexLevelParameteriv(0, 0, GL_TEXTURE_WIDTH, &texWidth);
    ^~~~~~~~~~~~~~~~~~~~~~~~
    glTextureParameteri
make[2]: *** [CMakeFiles/mupen64plus-video-GLideN64.dir/build.make:947: CMakeFiles/mupen64plus-video-GLideN64.dir/Graphics/OpenGLContext/GLSL/glsl_SpecialShadersFactory.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:74: CMakeFiles/mupen64plus-video-GLideN64.dir/all] Error 2
make: *** [Makefile:84: all] Error 2
/home/pi/RetroPie-Setup/tmp/build/mupen64plus /home/pi/RetroPie-Setup
Removing additional swap
/home/pi/RetroPie-Setup
Could not successfully build mupen64plus - N64 emulator MUPEN64Plus (GLideN64/projects/cmake/plugin/Release/mupen64plus-video-GLideN64.so not found).

For the entire log I've attached the log file to this entry.
[Error log](https://github.com/RetroPie/RetroPie-Setup/files/3902332/Error.log)

Any help is greatly appreciated as I'm somewhat stumped as to how to rectify this issue for a successful build of Mupen64plus through the Retropie setup script.

cmitu, 2019-11-29:
The problem might be in the upstream project - they added some changes recently. Can you re-try the installation and see if you get the same error ?

unverbraucht, 2019-11-29:
I also had the setup abort on me yesterday, telling me that the mupen glide .so couldn't be found. I didn't see anything interesting in the log but I'd be happy to dig it out and upload it if it helps.

cmitu, 2019-11-30:
I just built `mupe64plus` and the error reported originally is gone. There's an additional error introduced by a recent commit, but it's really an upstream issue (see the https://travis-ci.org/gonetz/GLideN64/jobs/618767859 build log) which I'm sure it will be fixed.

closed 2019-11-30 by joolswills

kylepd49, 2019-11-30:
I’ve tried rebuilding Mupen64Plus, I’ve tried starting from scratch with the FKMS branch again and I still receive the same thing. Apparently someone felt as though this thread should be closed without explanation however.

Closed #2909<https://github.com/RetroPie/RetroPie-Setup/issues/2909>.

cmitu, 2019-11-30:
... and it's fixed in https://github.com/gonetz/GLideN64/commit/80f1d9d0215a771fb8f40917147271c5ad7eface, so now there shouldn't be a problem anymore.

joolswills, 2019-11-30:
I closed it because it's an upstream problem that would be fixed - it's not a retropie bug. But we are not supporting the dev branch yet either.

kylepd49, 2019-11-30:
Thanks for the quick response! Good to know, I’ll check that out. Thank you for explaining the reasoning behind closing the thread. Happy holidays to both of you.