RetroPie/RetroPie-Setup #2930: RPI4 - 64bit - MUPEN64Plus does not compile on fkms_rpi4 branch

rafal-prasal, 2019-12-22:
HW: RPI4 - 4GB
arm64_enable=1 in /boot/config.txt
uname: Linux raspberrypi 4.19.88-v8+ #1284 SMP PREEMPT Wed Dec 11 13:57:50 GMT 2019 aarch64 GNU/Linux

Message:
Could not successfully build mupen64plus - N64 emulator MUPEN64Plus
(mupen64plus-core/projects/unix/libmupen64plus.so.2.0.0 not found)

how to reproduce:
git clone --single-branch --depth=1 --branch fkms_rpi4 https://github.com/RetroPie/RetroPie-Setup.git
cd Retropie-Setup
sudo retropie_setup.sh
choose basic install and proceed

from logs:
../../src/device/r4300/new_dynarec/arm64/linkage_arm64.S: Assembler messages:
../../src/device/r4300/new_dynarec/arm64/linkage_arm64.S:80: Error: ARM register expected -- `mov w0,w1'

solution should be to force 32bit options on mupen64Plus by adding
HOST_CPU=....

dankcushions, 2019-12-22:
alternatively you could set HOST_CPU=aarch64 to use the aarch64 dynarec - https://github.com/mupen64plus/mupen64plus-core/blob/master/projects/unix/Makefile#L134

rafal-prasal, 2019-12-22:
the aarch64 gets autodetected and does not compile. for the time being the workaround i am proposing is forcing 32bit here.

dankcushions, 2019-12-22:
do you have the aarch64 compatible version of GCC? because raspbian doesn't include it, and aarch64 content won't compile as a result.

IMO we shouldn't tacitly support the experimental 64-bit version of raspbian with workarounds like this, as there would need to be many. i think a more exciting thing to do would be to actually support a rpi4_64 platform :) but maybe not quite yet!

joolswills, 2019-12-22:
Sorry but this is not something we are going to support at this time. Also you may find performance worse on aarch64 due to some emulators losing neon optimisations etc.

closed 2019-12-22 by joolswills

rafal-prasal, 2019-12-22:
Actually I took newest raspbian and moved with kernel to aarch64 one and noticed that mupen64 does not compile, so I pushed to find solution for it as I sometimes like challenges ;).

At this point i checked PS1, SNES. NES, and they work quite well. after adding this "workaround" I was able tu run also N64. now looking at DC ;)

From what i read the plan for RPI4 is to have aarch64 kernel and armvX userspace and then maybe go for 64Bits, but to be honest you need 4GB+ of ram in order to have real value added to it, so probably 32 bits will stay in the userspace forever ;) ~2-5y :)

rafal-prasal, 2019-12-23:
> 
> 
> Sorry but this is not something we are going to support at this time. Also you may find performance worse on aarch64 due to some emulators losing neon optimisations etc.

Exactly, this is what I am thinking about. Let me explain.

Back in the day there were purists that wanted everything to be 64bits and actually this was impossible, because the apps were not ready and there was no point because max ram for desktop was around 1G and 4G was limit for motherboards. Sounds familiar to the situation in RPI4, right?

So I was thinking, instead of pushing everything to aarch64 it would be much better to enforce 32bits where it makes sense and let it be 64bits where it really gives performance boost or makes emulator running.

NEON you mentioned is not available in aarch64, but it does not mean that there are no equivalent operations, which do work as quick as NEON, they will need to get recoded, and emulators will have to be ready, but it is another story.

When it comes to emulators, then

joolswills, 2019-12-23:
I don't disagree but I'd like to visit this after our initial rpi4 release :-)