RetroPie/RetroPie-Setup #2951: generic amolgic s9xxx boards

NunoCruzSW, 2020-01-07:
I have a tv box with armbian installed, debian, and I dont have a __platform that can be use to install on a s905x board, vero 4k and odroic-c2 have this kind o cpu but are demanding install dependencies that are not easy to resolve.

there's any way to create a installation for a generic s9xxx board ?

cmitu, 2020-01-08:
Creating a new platform is not that straighforward - as you've seen, even if the CPU/architecture is the same, the supporting distro and other factors are in play. 
At a minimum, you'd need to create a new platform, based on Armbian, and check if the core packages are installable. You can see how that was added for the [Vero4k platform](https://github.com/RetroPie/RetroPie-Setup/pull/2438/) in a previous PR.

joolswills, 2020-01-08:
I may have a board like this (I need to check). I will revisit our non RPI board support later this year though. But help is welcome. Thanks.