RetroPie/RetroPie-Setup #2961: Review default emulators for Pi4

dankcushions, 2020-01-21:
**SNES**
Current: [lr-snes9x-2010](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-snes9x2010.sh)
Proposed: [lr-snes9x](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-snes9x.sh)
Libretro fork of snes9x upstream. Most accurate/compatible, and runs fullspeed on pi4 (but not on a pi3). **Would suggest making the default for pi4**.

**Megadrive/Genesis**
Current: [lr-picodrive](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-picodrive.sh)
Proposed: [lr-genesis-plus-gx](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-genesis-plus-gx.sh)
Libretro fork of genesis-plus-gx upstream. Better audio, at least. Was previously agreed to be default for pi2 and up: https://github.com/RetroPie/RetroPie-Setup/issues/688 - might have been lost in a refactor.
**Would suggest making the default for pi2 and up.**

**Mastersystem**
**Gamegear**
As per the issue above, but I've not tested this yet.
**Would suggest making the default for pi2 and up.**

**N64**
Currently [lr-mupen64plus-next](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-mupen64plus-next.sh) is in the experimental packages. At this point I think it's mature enough to be in main packages for pi4, with the abandoned [lr-mupen64plus](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-mupen64plus.sh) retiring to optional. lr-mupen64plus should probably stay in main for Pi3 and previous, but it does seem to work better for Pi3 + mesa.
**Would suggest mupen64plus standalone remains default, but lr-mupen64plus-next promoted to 'main' for pi4 and pi3 mesa, with lr-mupen64plus retiring to 'opt'**

**Dreamcast**
WIP: [lr-flycast](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/libretrocores/lr-flycast.sh) seems better for Pi4.

**Saturn**
WIP: Investigate lr-yaba-sanshiro (and/or standalone) install script as a likely good default.

darksaviorx, 2020-01-21:
I've suggested adding lr-neo-cd in the past, and now that it's an official libretro core, I wonder if I can ask again. It's the only neogeo cd emulator that supports chd. A developer for fb-neo showed no interest in adding chd support. I've only tested on a pi3b+ and it runs full speed on that. 

I would also like to suggest adding lr-beetle-pce since it's the accurate version of lr-beetle-pce-fast. The accurate version makes Exile playable. The fast version makes the ants in the first dungeon invincible. Maybe other games are affected as well. It runs full speed on a pi4 at stock speeds.

dankcushions, 2020-01-22:
@darksaviorx thanks - updated the list!

> I compiled the beetle-psx dynarec branch by making the retropie beetle-psx script point to it. I'm not dev so not sure what's the issue. Runs and looks like crap. Probably due to this error message when a game loads: "Unable to find or open hardware renderer for frontend preferred hardware context. Falling back to software renderer."

not sure - possibly a build option. looking at https://github.com/libretro/beetle-psx-libretro/blob/14b59e9875147307fa74d5ad4b0952fcf38cc18d/libretro-common/include/libretro.h#L2177 maybe we need mesa 3.1 support to be part of the pi updates until libretro returns a viable GLES3 context but I've not looked into it!

SpaceAgeHero, 2020-01-22:
Perhaps BSNES could be considered as well:
https://www.youtube.com/watch?v=MXPdbjZhPlo
Would love to see it with the Widescreen stuff arriving.

twinaphex, 2020-01-22:
You should be able to compile it without HAVE_OPENGL=1 being defined. If HAVE_OPENGL and HAVE_VULKAN are both not defined, then only the software renderer should remain.

And yes, the OpenGL renderer right now in Beetle PSX requires a high-ish subset of desktop GL, no GLES2/3 support. It'd be nice if somebody eventually comes up with codepaths for those however.

dankcushions, 2020-01-22:
> And yes, the OpenGL renderer right now in Beetle PSX requires a high-ish subset of desktop GL, no GLES2/3 support. It'd be nice if somebody eventually comes up with codepaths for those however.

ahh nuts. oh well, if the dynarec gets the software renderer to fullspeed it should still be an improvement on lr-pcsx_rearmed (which is software and has various compatibility issues).

quicksilver7837, 2020-02-01:
Not exactly an emulator but I think skyscraper could come out of experimental packages. I've been using it for over a year now and haven't had any issues with stability.

cmitu, 2020-02-05:
@quicksilver7837 can be added easily. Probably the main reason it's still in `experimental` is that installing from source was the better option with @muldjord adding new features and new versions at the time.

@dankcushions I remember testing `lr-mesen` on Pi4 and it wasn't fullspeed on Castlevania 3. I can re-test and try new games.

For **Gameboy** systems - add [Sameboy](https://sameboy.github.io/)'s Libretro core. Runs fine on Pi3/Pi4, has link cable emulation (2 screens side by side), supports RetroAchievements and features Super Game Boy (2) emulation. Upstream is very active and has very good Libretro integration.

darksaviorx, 2020-02-06:
@cmitu CV3 was one of the few games I tested with lr-mesen but feel free to verify. I re-tested it now and noticed no drops on stock speeds. Mesen's overclocking will require a pi4 overclock but that's optional. Hd packs are also optional but I haven't re-tested that in months. I faintly remember that would cut speed in half.

cmitu, 2020-02-06:
@darksaviorx @dankcushions I re-tested `lr-mesen` and it looks ok on the PI4. 
I must have been remembering it wrong, I also tested it a few times on a PI3 and on that platform it was not fullspeed.

dankcushions, 2020-02-06:
the dev of mesen says @ https://www.reddit.com/r/emulation/comments/65s9w3/comment/dgddny9
> The fact is, you'd be very hard pressed to find any emulation issues in Nestopia.

so for me maybe it’s best that this stays as an optional core, as i think the default should be the most performant core that gives meaningful accuracy, so you can have some overhead for run ahead, shaders, etc.

still, an install script seems worthwhile in any case. could have it disabled for pi1-2, and experimental for pi3.

quicksilver7837, 2020-02-10:
Should parallel-n64 be moved to experimental packages? It seems to seg fault or throw an illegal exception on just about every game and crash back to ES. This core had a lot of issues on my pi 3 as well.

joolswills, 2020-02-10:
Try setting core video plugin manually via core options. It might just need some better defaults.

quicksilver7837, 2020-02-10:
I set it to rice prior to your recent parallel commit. I have it set to auto currently but will try rice again. I'm out of town for a few days so will report back later.

quicksilver7837, 2020-02-17:
Is melonDS worth looking into as a replacement for drastic for NDS emulation? Drastic has some issues on the PI4 and we aren't likely to get any more updates at least until if/when drastic goes open source.

dankcushions, 2020-02-17:
Drastic is written for ARM so likely the only one ever to be viable.  feel free to try, though.

quicksilver7837, 2020-02-17:
Yes, sorry looks like I spoke out of turn. Some research shows that probably isn't a viable option on a pi.

darksaviorx, 2020-02-19:
I ran some more tests with the accurate version of Beetle PCE because I forgot arcade card games have much higher requirements. All are tested on stock speeds. 

Sapphire (arcade card game): Pi3b+ ~46fps, Pi4 59.92-60fps. 

The pi4 produces some frame drops but not low enough to produce scratchy audio. I don't mind if it's not the default, but It'd be great if it's in somewhere in retropie-setup. Thanks.

rileyinman, 2020-08-10:
Would definitely love to see sameboy as an option, given its feature set!

rileyinman, 2020-10-09:
It looks like a functional PR for Sameboy was created in #3083—I'm not sure why it was closed, but it could be a candidate for a merge.

cmitu, 2024-02-16:
I don't think this is relevant anymore, so closing.

closed 2024-02-16 by cmitu