RetroPie/RetroPie-Setup #2965: Unmet deps while installing sixaxis helper service for Shanwan/GASIA third-party controllers on Debian

casasfernando, 2020-01-31:
The installation for sixaxis helper service for Shanwan/GASIA third-party controllers fail due to unmet dependencies on Debian 9 and 10 (x86_64).

```
Log started at: Fri Jan 31 21:56:28 CET 2020

RetroPie-Setup version: 4.5.11 (ff4e2a2)
System: Debian GNU/Linux 10 (buster) - Linux retropie-x86 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u2 (2019-11-11) x86_64 GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Running action 'gui' for 'sixaxis' : Helper service to install & configure the latest drivers for official and third-party DualShock controllers (ps3controller replacement)

Note: For Shanwan/GASIA third-party controllers, enable third-party support in the configuration options.

To pair controllers, use the RetroPie Bluetooth menu, choose 'Register and Connect...', then follow the on-screen instructions.
= = = = = = = = = = = = = = = = = = = = =

= = = = = = = = = = = = = = = = = = = = =
Installing dependencies for 'customhidsony' : Custom hid-sony driver backported from kernel 4.15
= = = = = = = = = = = = = = = = = = = = =

Did not find needed package(s): dkms linux-headers-generic. I am trying to install them now.
Reading package lists...
Building dependency tree...
Reading state information...
Package linux-headers-generic is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source

E: Package 'linux-headers-generic' has no installation candidate
Could not install package(s): dkms linux-headers-generic.

Log ended at: Fri Jan 31 21:57:16 CET 2020
Total running time: 0 hours, 0 mins, 48 secs
```

The reason for the failure is that the package *linux-headers-generic* is not available in Debian, but only in Ubuntu.
Output from *apt list* in Debian 10:

```
pi@retropie-x86:~/RetroPie-Setup/logs$ apt list | grep linux-header

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

linux-headers-4.19.0-6-686-pae/stable 4.19.67-2+deb10u2 i386
linux-headers-4.19.0-6-686/stable 4.19.67-2+deb10u2 i386
linux-headers-4.19.0-6-all-amd64/stable 4.19.67-2+deb10u2 amd64
linux-headers-4.19.0-6-all-i386/stable 4.19.67-2+deb10u2 i386
linux-headers-4.19.0-6-all/stable 4.19.67-2+deb10u2 amd64
linux-headers-4.19.0-6-all/stable 4.19.67-2+deb10u2 i386
linux-headers-4.19.0-6-amd64/stable,now 4.19.67-2+deb10u2 amd64 [installed,automatic]
linux-headers-4.19.0-6-cloud-amd64/stable 4.19.67-2+deb10u2 amd64
linux-headers-4.19.0-6-common-rt/stable,stable 4.19.67-2+deb10u2 all
linux-headers-4.19.0-6-common/stable,stable,now 4.19.67-2+deb10u2 all [installed,automatic]
linux-headers-4.19.0-6-rt-686-pae/stable 4.19.67-2+deb10u2 i386
linux-headers-4.19.0-6-rt-amd64/stable 4.19.67-2+deb10u2 amd64
linux-headers-5.3.0-0.bpo.2-686-pae/buster-backports 5.3.9-2~bpo10+1 i386
linux-headers-5.3.0-0.bpo.2-686/buster-backports 5.3.9-2~bpo10+1 i386
linux-headers-5.3.0-0.bpo.2-all-amd64/buster-backports 5.3.9-2~bpo10+1 amd64
linux-headers-5.3.0-0.bpo.2-all-i386/buster-backports 5.3.9-2~bpo10+1 i386
linux-headers-5.3.0-0.bpo.2-all/buster-backports 5.3.9-2~bpo10+1 amd64
linux-headers-5.3.0-0.bpo.2-all/buster-backports 5.3.9-2~bpo10+1 i386
linux-headers-5.3.0-0.bpo.2-amd64/buster-backports 5.3.9-2~bpo10+1 amd64
linux-headers-5.3.0-0.bpo.2-cloud-amd64/buster-backports 5.3.9-2~bpo10+1 amd64
linux-headers-5.3.0-0.bpo.2-common/buster-backports,buster-backports 5.3.9-2~bpo10+1 all
linux-headers-5.4.0-0.bpo.2-686-pae/buster-backports 5.4.8-1~bpo10+1 i386
linux-headers-5.4.0-0.bpo.2-686/buster-backports 5.4.8-1~bpo10+1 i386
linux-headers-5.4.0-0.bpo.2-all-amd64/buster-backports 5.4.8-1~bpo10+1 amd64
linux-headers-5.4.0-0.bpo.2-all-i386/buster-backports 5.4.8-1~bpo10+1 i386
linux-headers-5.4.0-0.bpo.2-all/buster-backports 5.4.8-1~bpo10+1 amd64
linux-headers-5.4.0-0.bpo.2-all/buster-backports 5.4.8-1~bpo10+1 i386
linux-headers-5.4.0-0.bpo.2-amd64/buster-backports 5.4.8-1~bpo10+1 amd64
linux-headers-5.4.0-0.bpo.2-cloud-amd64/buster-backports 5.4.8-1~bpo10+1 amd64
linux-headers-5.4.0-0.bpo.2-common-rt/buster-backports,buster-backports 5.4.8-1~bpo10+1 all
linux-headers-5.4.0-0.bpo.2-common/buster-backports,buster-backports 5.4.8-1~bpo10+1 all
linux-headers-5.4.0-0.bpo.2-rt-686-pae/buster-backports 5.4.8-1~bpo10+1 i386
linux-headers-5.4.0-0.bpo.2-rt-amd64/buster-backports 5.4.8-1~bpo10+1 amd64
linux-headers-686-pae/stable 4.19+105+deb10u1 i386
linux-headers-686/stable 4.19+105+deb10u1 i386
linux-headers-amd64/stable,now 4.19+105+deb10u1 amd64 [installed,automatic]
linux-headers-cloud-amd64/stable 4.19+105+deb10u1 amd64
linux-headers-rt-686-pae/stable 4.19+105+deb10u1 i386
linux-headers-rt-amd64/stable 4.19+105+deb10u1 amd64
```

I found the offending line in the file **/scriptmodules/supplementary/xpad.sh** and wrote a simple patch that should fix this for both Debian and Ubuntu:

```
diff --git a/scriptmodules/supplementary/xpad.sh b/scriptmodules/supplementary/xpad.sh
index 0a9c1b3..1097980 100644
--- a/scriptmodules/supplementary/xpad.sh
+++ b/scriptmodules/supplementary/xpad.sh
@@ -23,7 +23,7 @@ function _version_xpad() {
 function depends_xpad() {
     local depends=(dkms)
     isPlatform "rpi" && depends+=(raspberrypi-kernel-headers)
-    isPlatform "x11" && depends+=(linux-headers-generic)
+    isPlatform "x11" && depends+="(linux-headers-"$(uname -r)")
     getDepends "${depends[@]}"
 }
 
```

Thanks.

joolswills, 2020-02-01:
Please can you test this

https://github.com/RetroPie/RetroPie-Setup/pull/2966

casasfernando, 2020-02-01:
> Please can you test this
> 
> #2966

Thanks for looking into this so quickly.
I will test your patch and report back.

casasfernando, 2020-02-01:
Tested the patch and it works great.

Hopefully it will get merged soon.

Thanks!

casasfernando, 2020-02-02:
Closing this issue as I saw that the patch was merged.
Thanks for being so quick into fixing this!

closed 2020-02-02 by casasfernando