RetroPie/RetroPie-Setup #2973: Amiberry - Use new cmd line parameter instead of config file

midwan, 2020-02-03:
I've added a new command line option in Amiberry 3.1.x which allows you to select a QuickStart Model easily (that sets all configuration options accordingly for that model).

You can use `-model=<model>` where `<model>` is `A500` or `A1200` for example, to quickly select such a configuration.

Using this method, you don't need those `rp-a500.uae` and `rp-a1200.uae` files at all, and additionally you ensure that any new features will be used automatically (e.g. new defaults that might be added as options), which wouldn't always be the case with config files.

You can see the full list of commandline paramters and their syntax by using the `-h` or `--help` option as well.

joolswills, 2020-02-03:
Thanks. I don't think this would replace the need for those files as I find them useful to customise (and users can easily add more presets if needed. I'll have a think, but I may not change this - but users are free to use this if they prefer.

midwan, 2020-02-04:
No problem from my side, just wanted to inform you of this option as well. You can do with it as you please, depending on your demands :)

Feel free to close this issue if it's no longer relevant.

closed 2022-05-04 by midwan