RetroPie/RetroPie-Setup #3014: PCSX2: use AppImage instead of distro packages

casasfernando, 2020-03-10:
This is not really an issue but an enhancement instead.
Recently an AppImage build for PCSX2 became available (https://github.com/ferion11/PCSX2_Appimage). Using an AppImage instead of the Linux distro packages is far better for a couple of reasons:

- Linux distro independent (it will work on Debian, Fedora, Ubuntu, etc)
- The AppImage already contains all the dependencies to run the application.
- No need to enable 32bit architecture on 64bit systems.
- Installation is as simple as downloading the AppImage file and give it execution permissions.
- Updating is as simple as downloading a new build and replace the previous one.
- No dependency on the Linux distro package maintenance (ie: Debian package seems to be not actively maintained) and version.

I did this manually, downloading the AppImage and integrating it as another system in RetroPie and so far so good. No issues at all.

Would you consider adding this as an option for PCSX2 installation in RetroPie setup script as an alternative to the traditional installation using the Linux distro packages?

Thanks for reading.

cmitu, 2023-03-29:
Not at the moment (and it seems that the AppImage project has been closed). The upstream `pcsx2` project provides an AppImage themselves, so if we decide to use it, we can use their version.

The `pcsx2` package in RetroPIe has been updated recently and will work on 64bit hosts. For older Debian/Ubuntu version, the latest (at the time) compatible version is used.
Updates are done from the `daily` PPA for Ubuntu, so it should be fairly up-to-date.

closed 2023-03-29 by cmitu

casasfernando, 2023-03-29:
Thanks for looking into this.
Indeed a lot has changed since I opened this issue and I’m happily using the official pcsx2 AppImage for a couple of years.