RetroPie/RetroPie-Setup #3107: emulationstation black/blank screen

avalkauskas, 2020-04-19:
I have a RaspberryPi4 model b 4 gigs with latest Rasbian installed.
I've installed Retropie and made sure to boot in console mode so not to get the "could not get egl display" error. Now typing in "emulationstation" I get a black screen. I've tried different resolutions/screens with no progress. I also tried with and without USB controllers plugged in. This is the first time we're setting up RetroPie so any help would be appreciated. I know it's not supported on Rasp v4 yet, so I've changed the config file to Rasp v3.

ps. I'm also doing everything via installs rather than flashing images since the 4" screen that came with my Pie had a starting hpoint of flashing an image with the driver.

joolswills, 2020-04-19:
Please use the retropie forum for support.

closed 2020-04-19 by joolswills

avalkauskas, 2020-04-19:
They don't want raspberry 4 questions