RetroPie/RetroPie-Setup #3108: DS4 motion control in the latest versions cannot be disabled?

fastearle, 2020-04-20:
Ok, 4.5.1 retropie with ds4 controllers connected via dongle. In many games, the motion control takes over and will not allow you to allocate controls other than the motion control which sucks... The method of disabling this in advmame didn't work. This is the only area with an issue... And had been ongoing. The controller in advmame ads a controller and es sees this. Hence it sees two controllers instead of one .. this is an issue since you start in a different emulator, it wants the kb to be primary controller and not your ds4... Very frustrating.... Any help would be appreciated...

cmitu, 2020-04-20:
Please use the forum for support, as noted in the issue template.
Duplicate of https://github.com/RetroPie/RetroPie-Setup/issues/2436

closed 2020-04-20 by cmitu

fastearle, 2020-04-20:
This solution is no longer viable and does not fix the issue in the current
system...4.5.1... With the most current retroarch, as I stated in the
post...

Thanks