RetroPie/RetroPie-Setup #3111: Packages are missing dependencies

theqaninja, 2020-04-25:
Could not get on the forum was able to successfully downloand this onto my linux PC before with no issues. Am now getting the following error since the latest code push?
How does this get fixed?

Please see following error message below:

nathan@nathan-MF-001:~/RetroPie-Setup$ sudo ./retropie_setup.sh
Did not find needed dependencies: gcc g++ build-essential python3-pyudev. Trying to install them now.
Hit:1 http://us.archive.ubuntu.com/ubuntu bionic InRelease
Reading package lists... Done         
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

The following packages have unmet dependencies:
 g++ : Depends: g++-7 (>= 7.3.0-12~) but it is not going to be installed
       Depends: gcc-7 (>= 7.3.0-12~) but it is not going to be installed
 gcc : Depends: gcc-7 (>= 7.3.0-12~) but it is not going to be installed
E: Unable to correct problems, you have held broken packages.
Unable to install packages required by /home/nathan/RetroPie-Setup/retropie_packages.sh - Could not install package(s): gcc g++ build-essential python3-pyudev.
nathan@nathan-MF-001:~/RetroPie-Setup$

cmitu, 2020-04-25:
This is not a RetroPie issue, fix your Ubuntu package manager. As said in the issue template, you should use the forum, but it looks like you've been booted off because of post spam.

closed 2020-04-25 by cmitu

theqaninja, 2020-04-26:
Lol Calm down Cristina Mitrana, 😂 I have several other accounts and have
already you spammed you other ways. Retropie is not even the best front end
anyway. Do better for the retro community. You owe it to them.