RetroPie/RetroPie-Setup #3117: BOX86 support!!!

Askmewho, 2020-04-29:
Hi guys!! now than retropie use mesa you should add BOX86 as an experimental package (also an updater from source as usual bc it's on heavy development)
I have several tests on this emulator and it's getting better and better.

[https://www.youtube.com/watch?v=xW6irIWN-Nk&t=222s](url)

[https://www.youtube.com/watch?v=gSK9zbl9MyI&t=1719s](url)

and an all list here

[https://www.youtube.com/watch?v=gSK9zbl9MyI&list=PLhgmBihZpcBxWacPE05P-tN_ZuPBy8cxh](url)

Dagnillo, 2020-05-02:
I agree, would love to see this added

Askmewho, 2020-06-23:
Box86 seems to run great on wayland too. So the non x11 should not be an obstacle to have it on retropie.

Askmewho, 2020-06-25:
Box86 running on wayland on rk3399

https://youtu.be/a1NFvm22wFI

Askmewho, 2020-06-26:
I tested weston on retropie official beta image with box86 and it's running great aside major glitches on full screen, something it's wrong on mesa upstream or weston package.

cmitu, 2020-06-26:
@Askmewho Please don't use this issue tracker as a forum. If you wish to discuss this more, use the forum or submit a PR.

closed 2020-06-26 by Askmewho