RetroPie/RetroPie-Setup #3142: Sega Controller Setup for games like Madden

Patriot15, 2020-05-15:
Hey everyone,

I've been using retropie for a year or so now and I've come across a situation that I'm not sure there's a solution for.  In sports games like Madden 95, you have a playbook and select plays by using specific controller buttons.  However, due to the unusual nature of setting up a Sega Megadrive/Genesis (C is A, A is Y, etc), it becomes almost unplayable.  You see a play that can be selected labeled "A" but if you press the A button on the controller, you end up selecting a different play... Short of creating a cheat sheet to look at in order to select a play, is there anyway to make sense of the buttons within the emulator or the programming of the controller?

I appreciate any help you can offer.  Thanks everybody.

cmitu, 2020-05-15:
Please use the forum for support, the issue tracker is for bug reports.

closed 2020-05-15 by cmitu

Patriot15, 2020-05-15:
Where is the forum? I cannot find anything on here labelled forum or similar.

joolswills, 2020-05-15:
It was detailed in the text you removed when opening an issue.

Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

This includes edit suggestions for the wiki. There are more people to help on the forum.

Once a problem has been verified on the forum, an issue can be opened here.

Please remove this text before posting.