RetroPie/RetroPie-Setup #3157: Redream update fails due to bad download (wget)

fczuardi, 2020-05-30:
After some retropie updates I ended up losing my redream emulator. 

The error I got was one along the lines of :

```
Error running 'tar -xv -z -C /opt/retropie/emulators/redream' - returned 2
```

The full log is:

```
Log started at: Sat 30 May 16:57:17 -03 2020

RetroPie-Setup version: 4.6.1 (578f920)
System: Raspbian GNU/Linux 10 (buster) - Linux retropie 4.19.118-v7l+ #1311 SMP Mon Apr 27 14:26:42 BST 2020 armv7l GNU/Linux

= = = = = = = = = = = = = = = = = = = = =
Installing 'redream' : Redream Dreamcast emulator
= = = = = = = = = = = = = = = = = = = = =

gzip: stdin: unexpected end of file
tar: Child returned status 1
tar: Error is not recoverable: exiting now
Error running 'tar -xv -z -C /opt/retropie/emulators/redream' - returned 2

Log ended at: Sat 30 May 16:57:29 -03 2020
Total running time: 0 hours, 0 mins, 12 secs
```

Digging up on the [scriptmodules/emulators/redream.sh](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/emulators/redream.sh#L20) and [scriptmodules/helpers.sh](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/helpers.sh#L1030) files I noticed that the expected downloaded file from wget that is piped to tar was probably failing...

So I tried this on the terminal:

```
wget -O- "https://redream.io/download/redream.aarch32-raspberry-linux-latest.tar.gz" 
```

For which the output was:
```
--2020-05-30 17:07:52--  https://redream.io/download/redream.aarch32-raspberry-linux-latest.tar.gz
Resolving redream.io (redream.io)... 97.107.129.89
Connecting to redream.io (redream.io)|97.107.129.89|:443... connected.
ERROR: The certificate of ‘redream.io’ is not trusted.
ERROR: The certificate of ‘redream.io’ has expired.
```
I don't know if this is an issue with my particular network or the redream.io webserver, the fact is that `curl -O` for the same url works here, or wget using `--no-check-certificate`... so there is a workaround I guess :)

cmitu, 2020-05-31:
This is a deeper issue, triggered by the expiration of a [Root CA ](https://support.sectigo.com/articles/Knowledge/Sectigo-AddTrust-External-CA-Root-Expiring-May-30-2020). Wget uses `gnutls` for SSL, which is affected by this [bug](https://gitlab.com/gnutls/gnutls/-/issues/1008).

`curl` uses OpenSSL, which doesn't exhibit this problem.

inolen, 2020-05-31:
Thanks for the heads up, this should now be fixed.

cmitu, 2020-05-31:
@inolen thanks for the fast fix. @fczuardi can you try now to install ?

closed 2020-06-01 by cmitu