RetroPie/RetroPie-Setup #3160: Lock emulationstation version to 2.9.0 if GPU doesn’t support OpenGL 2

mrsilver76, 2020-06-05:
If you attempt to run a version of EmulationStation later than 2.9.0 on a GPU which doesn’t support OpenGL 2, then it fails to boot. 

The workaround is to modify `$HOME/RetroPie-Setup/scriptmodules/supplementary/emulationstation.sh` and set the version being installed to 2.9.0. See details [here](https://retropie.org.uk/forum/topic/25720/emulationstation-white-screen). 

Could we get `emulationstation.sh` to automatically check for OpenGL 2 and lock the version of EmulationStation to 2.9.0 if it’s found to be unsupported?

cmitu, 2020-06-05:
IMHO, downgrading the EmulationStation version _automatically_ would create confusion for users. As EmulationStation will move forward, new features added will not show up for those users and the question will arise why their installation is incomplete/old or the version shows an old version.

EmulationStation requires a GL2 capable card now, using it on a GL1 only GPU on a previous version is a decision that I think should be left to the the user, so they are aware they're using an old and - in the future - possibly unsupported version.

mrsilver76, 2020-06-05:
I’m not sure “downgrading” is the way I’d argue it - it’s more about installing a version of EmulationStation that works versus installing a version that doesn’t and never will. 

Right now you can sit through several hours of compiling only for EmulationStation to fail to boot and walk away with the assumption that RetroPie-Setup is broken.

If it’s not something that will be considered, it might be nice to at least warn people before they start that they are going to end up with something that won’t work.

cmitu, 2020-06-05:
> I’m not sure “downgrading” is the way I’d argue it - it’s more about installing a version of EmulationStation that works versus installing a version that doesn’t and never will.

Downgrade is , by definition, is installing an older version.

> If it’s not something that will be considered, it might be nice to at least warn people before they start that they are going to end up with something that won’t work.

That would be something to consider, if you can find a way to properly detect if the main GPU is able to support GL2 or higher.

tomaz82, 2020-06-05:
The thing here is, ES needs ( and always has ) ARB_texture_non_power_of_two.
What puzzles me is how this has ever been able to run on a GL1 card.

2.9.0 used the old renderer where SDL2 was never told which GL version it should use, so it just picked whatever was available ( GL1 in your case ) which should have looked broken due to missing ARB_texture_non_power_of_two.

The new renderer specifically asks for GL 2.0 because that's when ARB_texture_non_power_of_two was introduced.

tomaz82, 2020-06-05:
Upon further investigation, it seems that ARB_texture_non_power_of_two was required from GL 2.0, it may exist even on GL1.x cards, once I find out when it was introduced as an extension I'll back the PC GL renderer to 1.x instead of 2.0 ( just need to solve the x )

tomaz82, 2020-06-05:
Man, reading GL specifications sure is fun.. but at least now we know that ARB_texture_non_power_of_two was introduced in GL 1.5, so I'll recode the GL renderer back to that later today.

mrsilver76, 2020-06-05:
> > I’m not sure “downgrading” is the way I’d argue it - it’s more about installing a version of EmulationStation that works versus installing a version that doesn’t and never will.
> 
> Downgrade is, by definition, installing an older version.
> 

Sure don't disagree - but from a users POV you can't really downgrade from something you never had access to in the first place. If I never had a BA first-class ticket, they cannot downgrade me to cattle class. Likewise, it's not much of a downgrade if you can allow me to run ES vs me not being able to.

> That would be something to consider, if you can find a way to properly detect if the main GPU is able to support GL2 or higher. We can add this requirement to the docs, but most users will not read that.

`glxinfo | grep "OpenGL version string"` reports "1.4 Mesa 19.2.8" for me, I don't know how consistent this is across various platforms mind you so I'd defer to someone more knowledgeable.

mrsilver76, 2020-06-05:
> Man, reading GL specifications sure is fun.. but at least now we know that ARB_texture_non_power_of_two was introduced in GL 1.5, so I'll recode the GL renderer back to that later today.
> 
> @mrsilver76 what card are you running, and which GL versions does it support?

I'll be honest, I didn't understand much of what you said :)

I'm running a Samsung NC10 (Atom N270) netbook. If you want me to run some commands and paste the output, then please let me know!

tomaz82, 2020-06-05:
That notebook uses an onboard  Intel GMA 950 which apparently only does OpenGL 1.4, so I'll drop ES to use 1.4 and require the ARB_texture_non_power_of_two ( which the Intel GMA 950 supports )

tomaz82, 2020-06-06:
https://github.com/RetroPie/EmulationStation/pull/662 adds back a GL 1.4 renderer

mrsilver76, 2020-06-06:
Me and my ancient Stone Age GPU says thank you :)

closed 2020-06-26 by cmitu