RetroPie/RetroPie-Setup #3161: Problems with netplay guest

alessandropaggi, 2020-06-07:
I've posted many times on the forum about this issue, but I've still got no answer so here I am. I can't connect to netplay rooms with Retropie despite running the same core and the same rom (same CRC) of the host. Every time I try to connect to a netplay room I get the the messages:
"Compatible content found"
"Couldn't find a suitable core or content file, load manually"
I've also ran this test: I am running Retropie on a Pi4 and Retroarch on my laptop (same Retroarch version, same core, same rom). I did not perform any rom scan on neither system (see below), I was just launching the same rom on the same emulator. On the host side I went in Retroarch menu Netplay->Host->Start Netplay Host. On the guest side I went in Retroarch menu Netplay->Refresh Room List and selected the host room.

- Retropie is able to serve as a netplay host, that is, I was able to join the Retropie netplay host room as a guest from Retroarch running on the laptop.

- Retropie cannot join netplay as a guest, that is, when I try to join the host room of Retroarch running on the laptop, I get the messages shown before. This also happen on any netplay host room I see in the room list.

I also tried installing Retropie on an Ubuntu virtual machine, obtaining the same results.
I don't know if this related, but I also tried to create a playlist by scanning the rom directory, by going in Retroarch menu Main Menu->Load Content->Playlist->Scan Directory, selecting the rom directory and then Scan This Directory. On Retroarch running on the laptop this correctly created a playlist with the roms found in the directory, while Retropie is not able to find any content.

cmitu, 2020-06-13:
I managed to reproduce this, but it's not fully clear why it happens. The content is matched based on CRC (so I think a playlist won't help), but if fails with 
```
[...]
[INFO] [CONTENT LOAD]: CRC32: 0xc3508f7e .
[INFO] [Lobby]: Testing CRC matching for: C3508F7E|crc
[INFO] [Lobby]: Current content CRC: C3508F7E|crc
[INFO] [Lobby]: CRC match C3508F7E|crc with currently loaded content
[INFO] [Lobby]: Couldn't find a suitable core
```
which is puzzling, since the core is already loaded (!). The error message is from https://github.com/libretro/RetroArch/blob/01bd76405fc1441c200de54f5cfe920c6da44fcd/tasks/task_netplay_find_content.c#L145, but tracing back to see where `state->core_path` is set didn't show anything in particular.

The error shows up even with a vanilla RetroArch configuration (i.e. no RetroPie specific configs).
One thing that seems to work is connecting directly to the remote netplay host from the RetroPie guest, but that has a limited utility, since it requires a keyboard.

@alessandropaggi in your test, the RetroArch host on the PC is from another RetroPie installation or is it an install downloaded from the Libretro site ?

alessandropaggi, 2020-06-13:
@cmitu The host is running RetroArch downloaded from the Libretro site, which just works out of the box. I also set up a RetroPie running on a virtual Ubuntu machine, and it fails connecting to the RetroPie on the Pi4, since none of the can act as a guest. The RetroPie running on the virtual Ubuntu machine works as a host for RetroArch running on the laptop but cannot join as a guest, exactly like the RetroPie running on the Pi4.

alessandropaggi, 2020-06-13:
I don't know if this is related, but after loading a core on RetroArch running on the laptop if I go in the RetroArch Menu `Main Menu`->`Load Content`->`Playlists`->`Manual Scan`->`Core` I can see, besides `Unspecified`, the core that I just loaded. If I do the same thing on RetroPie after loading any rom I can only see `Unspecified`. It looks like RetroPie does not know the core it just loaded.

cmitu, 2020-06-13:
That's probably because there's no playlist associated - the playlist contains the core info.
``` json
   {
      "path": "/home/pi/RetroPie/roms/pc/Quake.conf",
      "label": "",
      "core_path": "/home/pi/src/c1/dosbox_core_libretro.so",
      "core_name": "DOSBox-core",
      "crc32": "",
      "db_name": ""
    },
```

It may be that RetroArch tries to re-load the core, but since the cores are not installed where RetroArch expects them to be, it errors out.

Did this process ever worked for you or is it something that's been - relatively - changed recently ? I did a test with 1.8.5 (as client, from RetroPie) and I get the same behavior.

alessandropaggi, 2020-06-13:
I think that I was able to create playlists with on older RetroPie versions, but I was never able to join netplay.

cmitu, 2020-06-13:
There is a workaround for this, just by adding the cores to the location where RetroArch expects them:
* copy the core `.info` file(s) from https://github.com/libretro/libretro-core-info to `$HOME/.config/retroarch/cores`
* symlink the core `.so` file to the same location - for instance `fceumm`:
```
ln -s /opt/retropie/libretrocores/lr-fceumm/fceumm_libretro.so $HOME/.config/retroarch/cores
```

alessandropaggi, 2020-06-13:
Ok, this changed something. With the symbolic link to the core in the same directory of the core infos I am able to create playlists. RetroPie is still able to serve as a host. When I try to join netplay RetroPie loads the correct content, but it always shows the message
`Compatible content found`
`Failed to initialize netplay`
This happens when trying to connect to RetroArch running on the laptop and to every netplay host shown in the netplay room list.

cmitu, 2020-06-14:
Haven't tried all the situations, but in my tests I always have the (same) content loaded when starting the netplay, both on host and client. I didn't use a playlist at all. 

Enable verbose logging on the client and and post the debug messages from the client (not the whole log, just the portion regarding netplay initialization).

alessandropaggi, 2020-06-14:
This is the log of RetroArch running on the laptop successfully joining the RetroPie host
```
[INFO] CRC32: 0xb19ed489 .
[INFO] [Lobby] Testing CRC matching for: B19ED489|crc
[INFO] [Lobby] Current content crc: B19ED489|crc
[INFO] [Lobby] CRC match B19ED489|crc with currently loaded content
[INFO] [Lobby] Loading core /Applications/RetroArch.app/Contents/Resources/cores/snes9x_libretro.dylib with current content
[INFO] [Netplay] connecting to 2.235.172.186:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[libretro INFO] Nonexistent Port (8).
[libretro INFO] Nonexistent Port (9).
[libretro INFO] Nonexistent Port (10).
[libretro INFO] Nonexistent Port (11).
[libretro INFO] Nonexistent Port (12).
[libretro INFO] Nonexistent Port (13).
[libretro INFO] Nonexistent Port (14).
[libretro INFO] Nonexistent Port (15).
[INFO] Connected to: "saccublenda"
[INFO] [netplay] You have joined as player 2
```
(I don't know if the `Nonexistent Port` entries have some meaning).
This is the log of RetroPie unsuccessfully trying to join the RetroArch host
```
[INFO] CRC32: 0xb19ed489 .
[INFO] [Lobby] Testing CRC matching for: B19ED489|crc
[INFO] [Lobby] Current content crc: B19ED489|crc
[INFO] [Lobby] CRC match B19ED489|crc with currently loaded content
[INFO] [Lobby] Loading core /home/pi/.config/retroarch/cores/snes9x_libretro.so with current content
[INFO] [Netplay] connecting to 2.235.172.186:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[ERROR] Failed to set up netplay sockets.
[WARN] Failed to initialize netplay.
```

cmitu, 2020-06-14:
The failures on the PI site look like network/connection issues, not related to RetroArch.

alessandropaggi, 2020-06-14:
Yes, but the same happens when the laptop is connected to another network:
laptop as a host
```
[INFO] CRC32: 0xb19ed489 .
[INFO] [Lobby] Testing CRC matching for: B19ED489|crc
[INFO] [Lobby] Current content crc: B19ED489|crc
[INFO] [Lobby] CRC match B19ED489|crc with currently loaded content
[INFO] [Lobby] Loading core /Applications/RetroArch.app/Contents/Resources/cores/snes9x_libretro.dylib with current content
[INFO] [Netplay] connecting to 2.235.172.186:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[INFO] Connected to: "saccublenda"
[INFO] [netplay] You have joined as player 2
```
RetroPie as a host
```
[INFO] CRC32: 0xb19ed489 .
[INFO] [Lobby] Testing CRC matching for: B19ED489|crc
[INFO] [Lobby] Current content crc: B19ED489|crc
[INFO] [Lobby] CRC match B19ED489|crc with currently loaded content
[INFO] [Lobby] Loading core /home/pi/.config/retroarch/cores/snes9x_libretro.so with current content
[INFO] [Netplay] connecting to 93.56.72.43:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[ERROR] Failed to set up netplay sockets.
[WARN] Failed to initialize netplay.
```

cmitu, 2020-06-14:
If I look at the log, I can see that it's the same (host) IP address used to connect for both cases:
* RetroPie as host
```
[INFO] [Netplay] connecting to 2.235.172.186:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[INFO] Connected to: "saccublenda"
```
* RetroPie as guest
```
[INFO] [Lobby] Loading core /home/pi/.config/retroarch/cores/snes9x_libretro.so with current content
[INFO] [Netplay] connecting to 2.235.172.186:55435 (direct)
[INFO] [netplay] Connecting to netplay host
[ERROR] Failed to set up netplay sockets.
```

This is not a LAN address - did you forward the netplay port (`55435`) on your router to go the RetroPie host ?

alessandropaggi, 2020-06-14:
> This is not a LAN address - did you forward the netplay port (`55435`) on your router to go the RetroPie host ?

Yes.

cmitu, 2020-06-14:
Then it can't possibly work - at least in this test - since RetroPie client cannot conenct to itself. Try using the LAN address to connect. It's a bit difficult to see in RGUI, but using the XMB or Ozone input driver I think you're able to see the IP address.

alessandropaggi, 2020-06-14:
I've set the respective LAN addresses and the result is always the same (RetroArch connects as guest, RetroPie does not). I've also tried disabling the port forwarding, and connecting the laptop to another network. The result did not change.

cmitu, 2020-06-14:
I can't reproduce this case - for me the LAN connection worked - with the aforementioned workaround.

alessandropaggi, 2020-06-14:
Can you please pastebin your `retroarch.cfg`? Maybe it's some setting that I can't figure out.

cmitu, 2020-06-14:
I have a stock RetroArch config - if you'd like to replicate it, just replace yours with the `retroarch.cfg.rp-dist` file.

alessandropaggi, 2020-06-14:
Still the same behavior. I guess I have some connection problems or router configuration to fix.
Thanks for your help.

alessandropaggi, 2020-06-15:
I did some test on netplay rooms and was finally able to join as a guest. I guess I was having some lan problems.

cmitu, 2020-06-15:
Glad you sorted out. 
We'll see if we can find a solution to the main problem - core (re)loading from a RetroPie guest.

cmitu, 2022-06-15:
@alessandropaggi this has taken more than needed, but it should be solved with #3549 and #3548.
I've been meaning to solve it a few times, but always got derailed with a half-hacked implementation and put it on the side.

But now, since RetroArch 1.10.1 will make - almost - mandatory the presence of `.info` files, I took to opportunity to find a simpler solution than the workaround.

I don't know if you can still test it (make sure to update RetroPie-Setup first and then the RetroArch package), but would be nice to have a 2nd confirmation.

alessandropaggi, 2022-06-16:
@cmitu Thanks for looking into it.
I'm a little busy with work at the moment, but I will gladly test it next week. Do you suggest to remove the symbolic links before testing it?

cmitu, 2022-06-16:
Yes, you should remove them and also clear the `.info` files. Updating `retroarch` will create back the `.info` files, symlinks shoulnd't be needed anymore.
Thank you.

alessandropaggi, 2022-06-19:
After removing the symbolic links and the `.info` files, I've updated RetroPie-Setup script and then the RetroArch.
The `$HOME/.config/retroarch/cores` gets populated with the `.info` files, however without the symbolic links to the `.so` files I'm not able to join a netplay room, as I keep getting the `Couldn't find a suitable core or content file, load manually` message, even if the same rom is loaded and running on the same core.
After recreating the symbolic links I am again able to join the netplay rooms.
Running as a host always works as before.

cmitu, 2022-06-20:
@alessandropaggi hm, can you also re-install `runcommand` ? After running a game, can you check what are the contents of `/dev/shm/retroarch.cfg` ?

alessandropaggi, 2022-06-20:
Ok, I've cleaned again the `$HOME/.config/retroarch/cores` directory, reinstalled `RetroArch` and `runcommand`.
The `retroarch.cfg` indeed contained the line

```libretro_directory = "~/.config/retroarch/cores"```

which I commented out. Trying to join a netplay room as a guest now just crashes retroarch going back to ES with this log

```Parameters: 
Executing: /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x/snes9x_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/Super Mario World (USA).sfc" --verbose --appendconfig /dev/shm/retroarch.cfg
[INFO] [Config]: Loading config from: "/opt/retropie/configs/snes/retroarch.cfg".
[INFO] [Config]: Appending config "/dev/shm/retroarch.cfg".
[WARN] [GameMode]: GameMode cannot be enabled on this system ("dlopen failed - libgamemode.so: cannot open shared object file: No such file or directory.") https://github.com/FeralInteractive/gamemode needs to be installed.
[WARN] [Config]: GameMode unsupported - disabling...
[INFO] RetroArch 1.10.0 (Git e9e85f3)
[INFO] === Build =======================================
[INFO] Capabilities:  NEON VFPv3 VFPv4
[INFO] Built: Jun 15 2022
[INFO] Version: 1.10.0
[INFO] Git: e9e85f3
[INFO] =================================================
[INFO] [Input]: Found input driver: "x".
[INFO] [Core]: Loading dynamic libretro core from: "/opt/retropie/libretrocores/lr-snes9x/snes9x_libretro.so"
[INFO] [Overrides]: Core-specific overrides found at "/home/pi/.config/retroarch/config/Snes9x/Snes9x.cfg".
[INFO] [Overrides]: No content-dir-specific overrides found at "/home/pi/.config/retroarch/config/Snes9x/snes.cfg".
[INFO] [Overrides]: No game-specific overrides found at "/home/pi/.config/retroarch/config/Snes9x/Super Mario World (USA).cfg".
[INFO] [Config]: Appending config "/home/pi/.config/retroarch/config/Snes9x/Snes9x.cfg".
/opt/retropie/supplementary/runcommand/runcommand.sh: line 1313:  6961 Segmentation fault      /opt/retropie/emulators/retroarch/bin/retroarch -L /opt/retropie/libretrocores/lr-snes9x/snes9x_libretro.so --config /opt/retropie/configs/snes/retroarch.cfg "/home/pi/RetroPie/roms/snes/Super Mario World (USA).sfc" --verbose --appendconfig /dev/shm/retroarch.cfg
```

This is the content of `/dev/shm/retroarch.cfg`:

```video_refresh_rate = "60"
video_fullscreen = "true"
video_fullscreen_x = "1920"
video_fullscreen_y = "1080"
libretro_directory = " /opt/retropie/libretrocores/lr-snes9x"
```

Uncommenting the `libretro_directory` line and putting back in `$HOME/.config/retroarch/cores` the `*.so` symbolic links brings the netplay working again.
I don't know if it is related in any way, but I have noticed that `retroarch.cfg` contains another line

```libretro_info_path = "~/.config/retroarch/cores"```

cmitu, 2022-06-20:
Thanks for testing. I haven't encountered any crash, but I wonder if it's because of an incorrect `libretro_directory`.

alessandropaggi, 2022-06-20:
> Thanks for testing. I haven't encountered any crash, but I wonder if it's because of an incorrect `libretro_directory` (the space at the beginning).

Yes, that's possible.

cmitu, 2022-06-20:
Are the host and client running the same core and ROM ?
Check if you haven't got `libretro_directory` set also in the system's `retroarch.cfg`.

alessandropaggi, 2022-06-20:
> Are the host and client running the same core and ROM ?
Yes.
> Check if you haven't got `libretro_directory` set also in the system's `retroarch.cfg`.
It is not.

cmitu, 2022-06-20:
What game are you using for testing ?

alessandropaggi, 2022-06-20:
Super Maro World for SNES. Note that the same rom and core works without cleaning the `$HOME/.config/retroarch/cores directory` and commenting the `libretro_directory = "~/.config/retroarch/cores"` line in `retroarch.cfg`.

cmitu, 2022-06-21:
I see. I've tried with _Super Mario World (USA)_ and the netplay works - I can connect to the host without the error (`Couldn't find a suitable core or content file, load manually`).

One difference may come from the `.cfg` files you have - can you upload both general `/opt/retropie/configs/all/retroarch.cfg` and `/opt/retropie/configs/snes/retroarch.cfg` ?
If the `snes` specific config is the default, then there's no need to upload it. Make sure you remove any user/passwords (netplay) from the `.cfg`.

alessandropaggi, 2022-06-21:
Thanks for taking the time to dig into in.
Here is the `/opt/retropie/configs/all/retroarch.cfg`: https://pastebin.com/vDQZcSGy
Here is the `/opt/retropie/configs/snes/retroarch.cfg`:

```
# Settings made here will only override settings in the global retroarch.cfg if placed above the #include line

input_remapping_directory = "/opt/retropie/configs/snes/"

#input_player1_joypad_index0
#input_player2_joypad_index1
#include "/opt/retropie/configs/all/retroarch.cfg"
```

I also use a core override, pasted here: https://pastebin.com/wZ1NF2cs

alessandropaggi, 2022-06-21:
Thanks anyway for taking the time to dig into it.
I will stick to the workaround for now.

cmitu, 2022-06-22:
@alessandropaggi thank you for testing.

Still can't explain why it's not working.
I've re-tested with `libretro_directory` set to default (`~/.config/retroarch/cores`) in both `retroarch.cfg` files, but its value is still overridden by `/dev/shm/retroarch.cfg` (which is generated by `runcommand` at runtime, based on the core name).

alessandropaggi, 2022-06-24:
Ok, I've tried with a fresh install, updated everything, and IT WORKS! I guess there's something wrong with my current configuration, but I will stick with it anyway until the bullseye image comes out. So long, and thanks for all the testing.