RetroPie/RetroPie-Setup #3192: How to remap controls in mupen64plus-GlideN64 on retropie x86

oldgregoire, 2020-07-10:
Apologies in advance if this is a basic question but I’m trying to remap the controls on a few games I’m running in mupen64plus-glideN64 on retropie x86 (4.6.4) on a PC running ubuntu 20.04 but didnt find any straight-forward answers online.  I’ve done core remap files on most my N64 games as they’re running in lr-mupen64plus-next or lr-parallel-n64 but see I cant remap mupen64plus-glideN64 in retroarch but figure there has to be a way to still do so - what am I missing?  If I add my parameters to the retroarch.cfg file in /opt/retropie/configs/n64 folder will that do the trick?  Is there a different file I need to add or adjust?  Or is this just not poss for this specific core?  Any pointers would be greatly appreciated.  Cheers!

cmitu, 2020-07-10:
Please use the forum for support questions, as explained in the issue template.

closed 2020-07-10 by cmitu

joolswills, 2020-07-10:
Following the text you removed would be a first step:

Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

This includes edit suggestions for the wiki. There are more people to help on the forum.

Once a problem has been verified on the forum, an issue can be opened here.

Please remove this text before posting.