RetroPie/RetroPie-Setup #3209: Amiberry next version: change of cmd line parameters syntax

midwan, 2020-08-04:
Hi guys,
In the next version of Amiberry, I'll be making a few changes that could be considered "breaking" ones.
I wanted to let you know earlier, so you have enough time to adapt your environment accordingly, and also handle any user requests that might come up (for example, if they haven't updated their RetroPie environment after you've implemented the changes).

Here are the 2 main areas that will be affected, along with the solutions:
1) The command line parameter syntax will be slightly changed, to comply more with POSIX standards. Where before we had options like `-autoload=<path to lha or cue filename>`, it will now become `--autoload <path to lha or cue filename>` instead.

The full list of parameters is below (taken from the usage output):

```
 -h                         Show this help.
 --help                     

 -f <file>                  Load a configuration file.
 --config <file>            

 -m <Amiga Model>           Amiga model to emulate, from the QuickStart options.
 --model <Amiga Model>      Available options are: A500, A500P, A1200, A4000 and CD32.

 --autoload <file>          Load a WHDLoad game or .CUE CD32 image using the WHDBooter.
 --cdimage <file>           Load the CD image provided when starting emulation (for CD32).
 --statefile <file>         Load a save state file.
 -s <config param>=<value>  Set the configuration parameter with value.
                            Edit a configuration file in order to know valid parameters and settings.

Additional options:
 -0 <filename>              Set adf for drive 0.
 -1 <filename>              Set adf for drive 1.
 -2 <filename>              Set adf for drive 2.
 -3 <filename>              Set adf for drive 3.
 -r <filename>              Set kickstart rom file.
 -G                         Start directly into emulation.
 -c <value>                 Size of chip memory (in number of 512 KBytes chunks).
 -F <value>                 Size of fast memory (in number of 1024 KBytes chunks).

Notes:
File names should be with absolute path.

Example 1:
amiberry --model A1200 -G
This will use the A1200 default settings as found in the QuickStart panel.
Additionally, it will override 'use_gui' to 'no', so that it enters emulation directly.

Example 2:
amiberry --config conf/A500.uae --statefile savestates/game.uss -s use_gui=no
This will load the conf/A500.uae configuration file, with the save state named game.
It will override 'use_gui' to 'no', so that it enters emulation directly.
```
This change is not yet committed to the `master` Amiberry repo, to avoid causing problems to the many users that might still be using the old syntax. :)

**Solution:** Update your launch script(s) accordingly, taking away the `=` and using `--` where necessary.

2) the config files are updated with new options (no real problem there, as there are defaults supplied), but also with some changes. Most notable is the Screen related changes, which are the following:
- (this doesn't apply to RetroPie on the RPI, since it's using the Dispmanx version which is full-window always) Since "real" full screen mode was implemented, which actually changes the monitor resolution as well, there are now 3 available screen mode options in Amiberry: windowed, fullscreen, full-window. The previously used "fullscreen" option, was actually full-window, so if you want that one, the config file should be updated to use it instead.
- The Screen Height value is updated to use the same standard as WinUAE (the upstream main emulator that Amiberry gets changes from). This means that old height values will no longer be correct, as they were using 50% of the actual screen height used (they were doubled internally). To minimize the problem, I've added a fix in the Amiberry code, which will automatically double the height value of a loaded config, if it's below a certain threshold.

**Solution:** The internal fix fo the Height value should cover most cases, but if anything strange happens, it's advised to recreate the affected config files from scratch, so they get the correct values.

I will of course update the relevant Amiberry wiki pages and announce these changes again through other channels, but you know people - not everyone reads the release notes, and they only come back if they run into problems... :)

joolswills, 2020-08-05:
Thanks for the info - When do you intend to merge these changes ?

midwan, 2020-08-05:
They are ready to be committed, so the only reason for a delay is to give distro maintainers some time to adjust. 
I would say, by the end of this week, if that works for you?

joolswills, 2020-08-05:
That's fine thanks.

midwan, 2020-08-07:
I've pushed the changes to `master` :)

joolswills, 2020-08-07:
Thanks. I'll make the changes to our code tomorrow.

monkaBlyat, 2020-08-07:
please enable amiberry on aarch64 
elif isPlatform "aarch64"; then
        platform="pi64"

Retro1968, 2020-08-08:
How do I have to change the entry in es_systems.cfg in RetroPie for the Amiga CD32?
I use .bin + .cue files. So far I have used this entry:

  <system>
    <name>amigacd32</name>
    <fullname>Commodore Amiga CD32</fullname>
    <path>/mnt/toshiba/roms/amigacd32</path>
    <extension>.adf .adz .cue .dms .ipf .lha .sh .uae .zip .ADF .ADZ .CUE .DMS .IPF .LHA .SH .UAE .ZIP</extension>
    <command>/opt/retropie/supplementary/runcommand/runcommand.sh 0 _SYS_ amiga %ROM%</command>
    <platform>amiga</platform>
    <theme>amigacd32</theme>
  </system>   

The path to amiberry is "/opt/retropie/emulators/amiberry/amiberry.sh"

I tried this one that didnt worked:

<command>./opt/retropie/emulators/amiberry/amiberry.sh --autoload %ROM%</command>

This didnt worked too.
<command>./opt/retropie/emulators/amiberry/amiberry.sh --cdimage %ROM%</command>

joolswills, 2020-08-08:
I've not implemented the changes yet. I'm working on it right now.

Retro1968, 2020-08-08:
So I don't have to make any changes myself, just wait until you've finished your work?

joolswills, 2020-08-08:
Yes. And then if you have issues, please use the retropie forum for help. Thanks.

Retro1968, 2020-08-08:
OK, Thanx.

joolswills, 2020-08-08:
The new launch script should now handle --config and --autoload correctly.

joolswills, 2020-08-08:
You will need to update RetroPie-Setup script and then amiberry or use the main update function in RetroPie-Setup

commit 7499ca7a (2020-08-08): amiberry - adjust amiberry launch script for argument changes - fixes #3209
 * use "--config " instead of "-config=" and "--autoload "
 * may adjust this further and split out amiberry launch to its own code rather than sharing the uae4arm launch code
closed 2020-08-08 by joolswills