RetroPie/RetroPie-Setup #3219: Amiberry - New version includes floppy drive sounds

midwan, 2020-09-05:
Hi guys,

The new version of Amiberry that just got merged into `master`, includes a new feature:
- Floppy drive sound support

The new feature is disabled by default, but can be enabled in the Sound Panel of the emulator's GUI.
When enabled, it will look for and use a few new files, that are added in the `data/floppy_sounds/` new directory:

```
drive_click.wav
drive_snatch.wav
drive_spin.wav
drive_spinnd.wav
drive_startup.wav
```

The files are included in the repo of course, but they'd need to be copied to the location you are installing Amiberry in.
The `data/floppy_sounds` directory should be accessible from the starting path in which the binary is located.

cmitu, 2020-09-06:
The installation copies the `data` source folder to `/opt/retropie/amiberry` (the installation folder for `amiberry` in RetroPie), so it should include the new sound files. I just tested with a source installation and the `.wav` files are copied to `/opt/retropie/emulators/amiberry/data/floppy_sounds/`, so it looks ok.

midwan, 2020-09-06:
@cmitu 
Great, then it should work without any issues. :) Feel free to close this issue if you don't need it.

cmitu, 2020-09-07:
Thank you for the heads-up.

closed 2020-09-07 by cmitu