RetroPie/RetroPie-Setup #3228: MAME (experimental) fails to compile on Ubuntu 20.04 mini/server

johnodon, 2020-09-18:
Ubuntu 18.04 works as expected.  20.04 no longer has python in the path.

= = = = = = = = = = = = = = = = = = = = =
Building 'mame' : MAME emulator
= = = = = = = = = = = = = = = = = = = = =

Removing additional swap
Adding 5075 MB of additional swap
Setting up swapspace version 1, size = 5 GiB (5321519104 bytes)
no label, UUID=a9d782a2-e629-4f3d-927e-4d78db137903
swapon: /home/pi/RetroPie-Setup/tmp/swap: swapon failed: Invalid argument
GCC 9 detected
makefile:1032: *** Python is not available in path.  Stop.
strip: 'mame64': No such file
Removing additional swap
/home/pi
Could not successfully build mame - MAME emulator (/home/pi/RetroPie-Setup/tmp/build/mame/mame64 not found).

joolswills, 2020-09-19:
Please use the forum first as detailed when opening an issue.

Python is usually in the path, so I don't think this is a retropie issue. Builds ok here on my x86_64 system with Ubuntu 20.04. Please provide details of system when you post on the forum (check the pinned topics). 

Thanks.

joolswills, 2020-09-19:
Actually this has been reported on the forum and is a known issue. Technically upstream needs a fix, but I believe @cmitu will submit a PR..

closed 2020-09-19 by cmitu

XD3l, 2020-09-19:
Unsubcribe

hifihedgehog, 2021-03-08:
As of today, I am still getting this error in a fresh and fully updated install of Ubuntu 20.04.

PR #3229 (2020-09-19): mame: use python3 to build, when available
Should fix the situation when there is not default 'python', but python3
is installed, like default installation of Ubuntu 20.04 LTS.

Fixes #3228.