RetroPie/RetroPie-Setup #3233: BPi M1 building has warning version stable-updates has bad syntax

Jir8taiwan, 2020-09-27:
Hi, whom may concern.

My Banana Pi M1 in Armbian 20.08.3  with kernel 5.8.11-sunxi
I am installing RestroPie-Setup Script 4.6.8 and have patched from the helper for BPi board identify (https://github.com/bernhard263/Bananapi-Retropie).

Due to many different questions by a new leaner on linux os in arm architecture.
I not only does many stupid failures in crash result, but also am slowly making up careful after many times of complete os reinstallation.
However, I detect new issues to search someone help in advanced.
Today, I look at execute ./retropie_setup.sh and immediate show off following error feedback
**dpkg: warning: version 'stable-updates' has bad syntax: version number does not start with digit**

I try to ignore to go on installing core packages in menuconfig selection.
And then, I also detect the first downloading line with the same error message in something.
I think that is an important to solve why I cannot to build complete up in the end.
Would someone please to advice how to find the real location to debug or solve it?
It will be helpful to me because I has costed many days on redo, reinstall, recompile and rebuild.

Thanks a lot in the beginning.

Jir8taiwan, 2020-09-27:
I have tried this post : https://retropie.org.uk/forum/topic/23984/solved-rpi4-sdl2-problem-during-installation-meanwhile-its-available-via-apt-get/11
However, it is still not sucessful and received the error message.
**Errors:
Could not successfully build sdl2 - SDL (Simple DirectMedia Layer) v2.x (/usr/local/src/RetroPie-Setup/tmp/build/sdl2/libsdl2-dev_2.0.10+5mali_armhf.deb not found).**

I think this is why I cannot go into command emulationstation with error:
```
Sep 27 20:26:56 lvl0:   Error initializing SDL!
        No available video device
Sep 27 20:26:56 lvl0:   Renderer failed to initialize!
Sep 27 20:26:56 lvl0:   Window failed to initialize!
```

joolswills, 2020-09-27:
Please follow instructions when opening an issue - in the text you removed:

>Please start a topic on the RetroPie forum before opening an issue - https://retropie.org.uk/forum/

>Once a problem has been verified on the forum, an issue can be opened here.

However the repository you reference is nothing to do with RetroPie and we don't currently support your board. For further help please use the forum as detailed.

closed 2020-09-27 by joolswills