RetroPie/RetroPie-Setup #3241: lr-snes9x.tar.gz.asc Not found during installation

ipochto, 2020-10-19:
Log file tail:
```
[cut]
= = = = = = = = = = = = = = = = = = = = =
Installing 'lr-snes9x' : Super Nintendo emu - Snes9x (current) port for libretro
= = = = = = = = = = = = = = = = = = = = =

Downloading https://files.retropie.org.uk/binaries/stretch/rpi3/libretrocores/lr-snes9x.tar.gz.asc ...
https://files.retropie.org.uk/binaries/stretch/rpi3/libretrocores/lr-snes9x.tar.gz.asc:
2020-10-19 11:10:08 ERROR 404: Not Found.
URL https://files.retropie.org.uk/binaries/stretch/rpi3/libretrocores/lr-snes9x.tar.gz.asc failed to download.

https://files.retropie.org.uk/binaries/stretch/rpi3/libretrocores/lr-snes9x.tar.gz.asc:
2020-10-19 11:10:08 ERROR 404: Not Found.

Log ended at: Mon Oct 19 11:10:08 CEST 2020
Total running time: 0 hours, 8 mins, 27 secs
```

joolswills, 2020-10-19:
Looks like there are some missing signatures for Stretch - I will sort it.

Note that Stretch support is being dropped soon, so it's a good idea to upgrade to a Buster based image.

ipochto, 2020-10-19:
Thanks.

>Note that Stretch support is being dropped soon, so it's a good idea to upgrade to a Buster based image.
I'd like to, but I've install RetroPie on the OSMC which still Stretch-based. As I know, they worked on migration to the Buster.

hissingshark, 2020-10-19:
As it happens OSMC transitioned to Buster yesterday.  The updates should roll out imminently.

joolswills, 2020-10-20:
Good news :-)

The package in question should now be fine. Other packages are in the process of being rebuilt - mostly experimental packages. I'll close this when they are uploaded.

closed 2021-01-12 by joolswills