RetroPie/RetroPie-Setup #3249: RetroArch doesn't save changed values that are already set in an included config

joolswills, 2020-11-02:
See https://retropie.org.uk/forum/topic/28346/configuration-via-rgui-partly-broken

If we have a custom config with

```
#include /opt/retropie/configs/all/retroarch.cfg
```

changing an option that is set in the config above in RGUI and using "Save current configuration" from RGUI will not save anything.

Bisected to https://github.com/libretro/RetroArch/pull/10524/commits/9eb84728050eb3ba371fd5af9c872ae1c577501b

For now I am going to revert this commit from v1.8.8, but will need revisiting for a proper fix.

joolswills, 2020-11-02:
https://github.com/RetroPie/RetroPie-Setup/pull/3250

Brunnis, 2020-11-03:
@joolswills I took the liberty of submitting an issue for this upstream: https://github.com/libretro/RetroArch/issues/11515

joolswills, 2020-11-04:
Appreciated. Thanks.

joolswills, 2021-01-12:
Just to update this - it was resolved in https://github.com/libretro/RetroArch/pull/11540 and will be part of Retroarch v1.9.1

closed 2021-06-27 by joolswills