RetroPie/RetroPie-Setup #3254: build error on 64bit os (dietpi)

gxitm, 2020-11-05:
Hardware:Raspberry Pi 3B
OS:  DietPi_RPi-ARMv8-Buster

log:
CC libretro-common/glsym/rglgen.c
CC gfx/drivers_context/vc_egl_ctx.c
gfx/drivers_context/vc_egl_ctx.c:49:10: fatal error: EGL/eglext_brcm.h: No such file or directory
 #include <EGL/eglext_brcm.h>
          ^~~~~~~~~~~~~~~~~~~
compilation terminated.
make: *** [Makefile:206: obj-unix/release/gfx/drivers_context/vc_egl_ctx.o] Error 1
~/RetroPie-Setup
Could not successfully build retroarch - RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators (/root/RetroPie-Setup/tmp/build/retroarch/retroarch not found).

joolswills, 2020-11-05:
Please use the forum as detailed when opening an issue. Also we don't support dietpi currently.

closed 2020-11-05 by joolswills

ahuffman, 2021-08-30:
I get the same error on 64-bit Ubuntu Server "20.04.3 LTS (Focal Fossa)".

Could not successfully build retroarch - RetroArch - frontend to the libretro emulator cores - required by all lr-* emulators (/home/ahuffman/RetroPie-Setup/tmp/build/retroarch/retroarch not found).
/home/ahuffman/RetroPie-Setup

Running a rpi 4b.