RetroPie/RetroPie-Setup #3256: Retropie-Manager is not installing

pandascripter, 2020-11-07:
When trying to install from the "exp" packages, the "retropie-manager" option is broken.
Apparently, there is a problem with the dependency python-dev.
Trying to run both python2-dev and python3-dev to solve the issue does not have a desirable effect.

joolswills, 2020-11-07:
Please use the forum first as detailed in the text you removed when posting.

closed 2020-11-07 by joolswills

hifihedgehog, 2021-03-08:
@joolswills, please reopen this issue because the user has addressed it on the forum. As far as a fix, you might consider this fork since it *may* work given that (quick code skimming) they are specifying Python 2 explicitly in the code which is one fix that would need to occur to get the old version working.

hifihedgehog, 2021-03-08:
This is the fix for Ubuntu 20.04. Here is the fix to the Makefile:

```
	virtualenv --python=/usr/bin/python2.7 .
	python2.7 -m pip install -r pip-requirements/basic.txt
	python2.7 -m pip install psutil
	python2.7 manage.py migrate
```
	
Prior to running the Makefile, pip will need to be installed in Python 2.7. You can do that simply by running the following:

```
curl https://bootstrap.pypa.io/pip/2.7/get-pip.py  --output get-pip.py
sudo python2.7 get-pip.py
```

hifihedgehog, 2021-03-09:
@joolswills Please consider integrating this fork as my fix was merged within hours so this should now work in Ubuntu 20.04 on 64-bit x86.