RetroPie/RetroPie-Setup #3260: There's a "retropie.pkg" file in the "$md_inst" folder after installing a scriptmodule

hiulit, 2020-11-17:
It's not an issue per se, because checking [the code](https://github.com/RetroPie/RetroPie-Setup/blob/3d0a4ed7673e3d54aa5332d2894534fe00727c13/scriptmodules/packages.sh#L499) I've found that it's intentional, but rather a question. Why is that? It wasn't there before, right?

I'm mentioning this because I have a scriptmodule that uses the files installed on `$md_inst` to configure the `emulators.cfg`, like this (maybe not the best way, tbh):

```
bin_files=($(ls "$md_inst"))
```
And this new file is getting in the way :P

Of course, I can easily bypass this issue.
I just wanted to know why is this new `retropie.pkg` file there and if it's going to be there in the future, so my "hack" to get the correct installed files will be permanent or it will be just temporary.

Thanks a lot for your time!

joolswills, 2020-11-17:
It's going to be there for the future, so you will need to adjust your code. Please use the forum first as requested when opening an issue.

closed 2020-11-17 by joolswills

hiulit, 2020-11-17:
Ok, thanks!
And sorry, I though that opening an issue here, as it was related to  RetroPie-Setup code, would be a better place. I'll keep that in mind.