RetroPie/RetroPie-Setup #3269: Enhancement - Feature Request

2play, 2020-12-10:
@joolswills  Hello dear friend!

If I can kindly request to consider a feature to have the video screen saver to play the video in a smaller part of the screen inside an "overlay". For example instead full screen video snap, have an old TV or similar that can be changed and adjusted as we can do in RetroArch custom overlays.

Just an idea.

Kind regards

Pi-Mania, 2020-12-10:
Great idea. Like it. screensaver video snaps playing in a static resizable area. 🤔👍

Traxis, 2020-12-10:
I also think it would be a great idea. If it's not possible to implement it, could it be done using the theme system? 🤔

joolswills, 2020-12-10:
Please use the forum as detailed when opening an issue - features can be discussed there first. Also this is the wrong repository.

closed 2020-12-10 by joolswills

2play, 2020-12-10:
I guess its ES. ok Forum post discussion.