RetroPie/RetroPie-Setup #3276: KDE Neon 20.04 rejected as "unsupported OS"

Taranchul, 2020-12-22:
Hi, Clyde from retropie.org.uk/forum here.

when installing following [these instructions](https://retropie.org.uk/docs/Debian/), RetroPie-Setup quits on KDE Neon 20.04 with this message:
```
Unsupported OS

Distributor ID: Neon
Description:    KDE neon User Edition 5.20
Release:        20.04
Codename:       focal
```

I consider this a bug since it had already been fixed in 2016, see #1800.

cmitu, 2020-12-22:
What's the output of
```
lsb_release -sidrc
```
on this distro ?

Taranchul, 2020-12-23:
```
Neon
KDE neon User Edition 5.20
20.04
focal
```

cmitu, 2020-12-23:
Can you try and see if the following modification to 'RetroPie-Setup/scriptmodules/system.sh` solves the problem:
``` diff
-        Ubuntu|neon|Pop)
+        Ubuntu|[N|n]eon|Pop)
             if compareVersions "$__os_release" lt 16.04; then
                 error="You need Ubuntu 16.04 or newer"
```

STPKITT, 2021-01-03:
I experienced this issue right now. I had it working perfectly fine on the 18.04 based KDE Neon.

@cmitu: I tried this, exchanged the red line you posted with the green one and put
if compareVersions "$__os_release" lt 16.04; then
in front of
error="You need Ubuntu 16.04 or newer"
but the result was exactly the same as before.

Regardless of this specific case with KDE Neon I would propose the option to show the unsupported OS message but let the user force installation anyway since judging by how many Ubuntu-based distributions are out there many users will potentially experience this issue the way retropie_setup.sh is written currently.

STPKITT, 2021-01-03:
I made it work by changing that one line to Ubuntu|Neon|neon|Pop).
Still option to force installation regardless is needed IMO.

cmitu, 2021-01-03:
@STPKITT Try this, should work.
``` diff
-        Ubuntu|neon|Pop)
+        Ubuntu|[Nn]eon|Pop)
             if compareVersions "$__os_release" lt 16.04; then
                 error="You need Ubuntu 16.04 or newer"
```

STPKITT, 2021-01-05:
I tried it and can confirm that it does work that way. So that change now just needs to find it's way into RetroPie proper.

cmitu, 2021-01-05:
Sould be fixed with https://github.com/RetroPie/RetroPie-Setup/commit/44c33c1ffa85ab6b9bb2ddab0090670552945e25

closed 2021-01-05 by cmitu

Taranchul, 2021-01-10:
Hello everyone,
sorry for the late answer. I somehow missed mitu's second reply 18 days ago, and when more people replied, RealLife™ held me firmly in its talons.

Thank you all for working on this in my absence, mitu's solution works fine on my Laptop with Neon 20.04. 👍