RetroPie/RetroPie-Setup #3305: Audio Settings Cause Controllers to Stop Working

mtoigo, 2021-02-04:
I wouldn't believe this bug either, but myself and others seem to confirm it and I'm happy to help troubleshoot. The two relevant forum posts are at:

- https://retropie.org.uk/forum/topic/29481/controls-randomly-locked-for-1-or-2-secs/11
- https://retropie.org.uk/forum/topic/29519/controllers-cut-out-briefly-in-n64-emulators/3

Since the last Retropie update 4.7.1 if you are using the 3.5mm jack on the raspberry pi (set in the audio settings in raspi-config), USB controllers will intermittently stop working for a few seconds. I went through a massive amount of troubleshooting in the second forum post above and couldn't solve it, but switching to HDMI audio output in raspi-config completely solves the issue.

joolswills, 2021-02-04:
I'd this is the case (I don't doubt you), you need to report this to Raspberry Pi OS. If I can reproduce I will try and help, but this will be a firmware /kernel issue

https://github.com/raspberrypi/firmware

mtoigo, 2021-02-04:
Thanks for pointing me in the right direction and I wasn't completely sure where to log it, but I'll add it there and we can close this out. I just want to save someone the 10 hours I burned on this problem :)

joolswills, 2021-02-04:
Sorry for your troubles. It's fine to leave this open. I'm interested in a solution. I haven't read the thread but I think I browsed it the other day but have you tried the latest kernel with `rpi-update` ?

This will need to be specified when reporting upstream btw. Sorry if this is already in the topics, I will read again when I get the chance.

cmitu, 2021-02-05:
You may want to try updating the kernel/firmware normally - i.e. without `rpi-update` - the latest beta kernel (that was available via `rpi-update`) has been promoted to stable and it's available through a normal update.

Taranchul, 2021-05-29:
> I just want to save someone the 10 hours I burned on this problem :)

Well, you just did! I noticed this problem some months ago, but didn't get around to look into it until now. Although it was rott42 in your first forum link who pointed out the audio jack as the cause of the problem, I found this issue report first. 👍

Did you report the problem to raspberrypi/firmware like Jools suggested? I couldn't find a similar issue report there.

I can confirm that this problem still exists in RetroPie 4.7 on my Pi 4 that was fully updated today (OS and RP). I would very welcome a fix since hdmi audio forces me to use a hardware audio splitter in my cab.