RetroPie/RetroPie-Setup #3306: Question: Microsoft repos being installed on raspbian

ricardoandren, 2021-02-04:
Does this privacy vulnerability also impact RetroPie? And are there any plans to add this repo to RetroPie aswell, I hope not.

joolswills, 2021-02-04:
Please can you provide a link?

ricardoandren, 2021-02-04:
https://www.raspberrypi.org/forums/viewtopic.php?t=301011

ricardoandren, 2021-02-04:
They are stealthly deploying microsoft vscode via raspberrypi-sys-mods

joolswills, 2021-02-04:
I've not read everything yet, but basically RetroPie images are based on Raspbian / Raspberry Pi OS, so we are impacted by their changes, and however they choose to resolve them. I doubt we will take a different route from whatever they decide but will check it out in detail.

ricardoandren, 2021-02-04:
Sooo your saying that unless it doesnt require a lot of changes made your keeping a capitalist-telemetry-ridden-privacy-violating repo from microsoft in your retro emulation OS, even if its just sending a text file back and forth? The server will still know you've used their repo, and also they will still know your IP.

joolswills, 2021-02-04:
We won't be doing anything on RetroPie in relation to this. It's not a RetroPie issue.

closed 2021-02-04 by joolswills