RetroPie/RetroPie-Setup #3313: Preparation for Amiberry v4.x

midwan, 2021-02-21:
Hi guys,

The next version of Amiberry is close to release now, so I would like to summarize a few key areas that are related to RetroPie, to ensure a smooth upgrade. I've run some tests with the latest version on the latest available RetroPie (on RPI4), and that's where this list comes from:

- Add Amiberry as an option for 64-bit platforms. Amiberry has 64-bit targets, but it appears that RetroPie-Setup does not offer them in the supported platforms. E.g. you can compile the 64-bit RPI4 version using the `PLATFORM=pi64` (for KMSDRM/X11) or `PLATFORM=pi64-dispmanx` (for Dispmanx). The full list of supported targets is in the Makefile. This was mentioned by some users who had a 64-bit distro installed, and added RetroPie on top - they couldn't install Amiberry from the menu.

- Update dependencies: with the next version of Amiberry, `libxml2` is no longer required. I've switched to a minimal included XML library, which is enough for our needs. One less thing to install, yay!

- Update `conf/amiberry.conf` file with the latest version. Some default values there are no longer correct with newer versions, and could lead to user frustration - e.g. the Default Height option, used to be 50% of the real height. This is no longer the case with newer versions, however whatever value is there will be used. This can result to having half the expected screen height showing in your games.

- Either remove the default config files, or update them. They contain deprecated config options, which again could cause issues when running with the newer version. Since you launch Amiberry from the console with parameters, the easiest way would be to remove them altogether and use the `--model` option to specify which model you want to emulate, instead. E.g. `--mpdel A500` (to start with an A500 config) or `--model A1200` to start with an A1200 config. You can always run the binary with the parameter `-h` or `--help` to see the full list of command line options. If for some reason you must have those config files there, then at least please recreate them using the latest version, to be sure that they are "clean".

You can test the upcoming version any time you want, by checking out the `dev` branch: https://github.com/midwan/amiberry/tree/dev - once this is released, it will be of course merged into the `master` branch and binaries will be made available, as usual (though I know you don't use the binaries I provide).

Feel free to ask anything you might need clarifying or if you need assistance with anything... :)
Let's make this release a great experience for everyone!

joolswills, 2021-02-21:
I need to look into this further. I may add the default systems as options on top of our customisable rp-a500 / rp-a1200 configs but resetting configs is tricky.

Users may have saved out a config for a specific ADF and may have many such configs.

It would be better if you had a new config for height or a new config flag which would affect how the height setting is evaluated imho. That way old configs can work, but when a user saves, they get an updated config.

I may leave this one up to users and document it as I don't see a way of doing this. When we change default configs we don't usually modify any existing. We just copy the default config to CONFIG.rp-dist

I wouldn't want to override any existing configs in case someone had customised them for a specific use.

joolswills, 2021-02-21:
Thanks for the update btw. Forgot to say that and appreciate all the development and letting us know.

midwan, 2021-02-22:
@joolswills 
I was referring to the default configs bundled with the installation of Amiberry. Of course changing pre-existing configs is outside the scope, and should be handled by the users themselves (as they may have modifications etc.). But the new ones that pop-up when you install Amiberry should either a) not exist at all, since they are not really needed if you use the `--model` parameter or b) be regenerated using the latest version, so at least they are clean from old/obsolete config parameters.

midwan, 2021-03-10:
@joolswills 
Maybe I misunderstood or perhaps I wasn't clear enough :)
I wasn't implying that you'd have to change existing configs, but rather stop providing those 2 default configs, since they are not really needed if they are meant for giving users a "Default A500" and a "Default A1200" config to start with. For those cases, you can use the command line parameters, instead.

Existing configs should be left alone and handled by the users (or the emulator itself, when you re-save them), of course.

I know it's not ideal to have config file changes, but in this case it was unavoidable I'm afraid. Older versions had incorrect settings, which I had to fix in order to synchronize with the upstream emulator (WinUAE). It shouldn't be an issue in the future, however.

midwan, 2021-03-24:
Today Amiberry reached "Release Candidate" status :)

Among the recent changes, there was one new directory added in the bundle: `abr`
It's optional, so it's OK if it's missing. If it's included, along with the 2 included XML files in it, it will be used by the emulator for recognizing ADF disk bootblocks, which can be shown in the relevant section in the GUI (Floppy Panel, Disk info button).

Otherwise there have been many updates, bug fixes and even some new features. Perhaps the most important one, is the addition of cycle-accurate 68000 emulation, for A500. This is disabled by default, to avoid any disruptions in existing configs/behavior, but also because it's slower and not really needed in most (>90%) cases. But it's there if you need it, now.

How are things from your side, regarding the upcoming version? All ready to go?

joolswills, 2021-03-30:
Nice job.

Nothing changed yet. I've not tested our shipped with minimal configs if they need changes.

I may lock to a commit until we are ready but apart from the new folder, I'm not sure if we can leverage the eg. Default a1200/a500 settings you have without breaking existing configs 

We currently map by default to rp-a500/rp-a1200 configs depending on name of file (eg if it contains AGA etc). I don't want to change existing behaviour.

The new version sounds good and I will test asap.

midwan, 2021-03-30:
@joolswills 
From the execution point of view, the change is minimal: you currently use a command line parameter to load a specific config, plus the selected game. In the new approach, you still use a command line parameter to pick a config, plus the selected game. The only difference is, there's no config file involved (the parameter triggers the default settings for the chosen Amiga model, like you'd select them from the "Quickstart" Panel).

This will ensure that any default options will be honored and automatically work with any new versions (whereas with old config files, very often you'd run into problems due to old/obsolete options or incorrect values).

If the users already have existing configs, of course we don't touch those. If they have problematic options in them, they'd need to be manually fixed by the user (by e.g. recreating that config and re-saving it).

But new installations should have a solution that doesn't cause problems, and the current configs do have some entries that are problematic (most common one: 50% of the actual screen height, which will lead to cropped output). That's why I came up with this approach in the first place.

Of course, I can only suggest what I think would be best. It's your choice in the end, as always. Just trying to minimize support tickets opening from users for known issues that could be avoided. So I will stop babbling about this here... :)

In other news, we're very close to the release of v4.0 - it might happen in a few days, if all goes well.

midwan, 2021-04-01:
@joolswills 
Amiberry v4.0 was released today! :)
https://github.com/midwan/amiberry/releases/tag/v4.0

Pre-compiled binaries are available as usual, in the Release page.

One user reported an issue with RetroPie, when using any configuration with JIT enabled:
https://github.com/midwan/amiberry/issues/784

However, this only seems to happen to me if I use RetroPie-Setup to compile from source. If I compile the binary myself, it works fine.
The two binaries also differ in size (RetroPie-Setup is about 14MB, manual compile is around 9.4MB).

Retro1968, 2021-04-02:
If you install the precompiled binaries via the RetroPieSetup, you get amiberry version 3.3. If you choose the source code you get the faulty version amiberry 4.0.

Retro1968, 2021-04-03:
With version amiberry 4.1 the error no longer occurs

nemo93, 2021-05-13:
@joolswills @midwan thanks again both for the interesting exchanges and for all the efforts to ensure Amiga still lives for a long time :-) In order to keep the ball rolling and to move forward to 4.x let me recap first briefly what the "wrapper" script is doing and then offer a potential solution.

The "wrapper" script named after `amiberry.sh` is being run whenever an amiga software is selected from the frontend (could be a .uae, .adf, .cue or .lha). Should a `.uae`, `.cue` or `.lha` be selected then Amiberry will be run straight to emulation via the `-G` flag. Should a `.adf` be picked up then Amiberry will be launched with one of Retropie-provided config file (either `rp-a500.uae` or `rp-a1200.uae` depending on the name of the software). Those files are automatically created by Retropie Setup (see [here](https://github.com/RetroPie/RetroPie-Setup/blob/cf1f3207a8e6fd76b4e6bb4f862a553c847a3939/scriptmodules/emulators/uae4arm.sh#L64-L86)). 

The issue we're discussing is for `.adf` only (does not concern WHD packages or CD32 images for instance).

Newer version of Amiberry (4.x and above) comes with a flag called `--model` that will configure the emulator in the best way for the Amiga model selected (ie. `--model A500`). Use of the above 2 configs file can be seen as obsolete. However for quality of life reasons and to keep the transition from Amiberry 3.x to 4.x as smooth as possible it is important to keep those files. At least to let the users decide to keep them whenever one decides to.

My proposition would be to add a new parameter to be passed to the "wrapper". Today there's only a single parameter `auto` to be passed to that script. We could perhaps add a second one named after `compat` (or `old` or `adf`) to enforce the use of config files. This will require users to change the `emulators.cfg` accordingly.

I'd modify the [script](https://github.com/RetroPie/RetroPie-Setup/blob/master/scriptmodules/emulators/uae4arm/uae4arm.sh#L49-L65) the following way. Sorry if it's not optimal code I just write it as I type, just to give the idea.
```
# if no config or auto provided, then look for rom config or choose automatically 
    if [[ -z "$config" || "$config" == "auto" || "$config" == "adf" ]]; then
        # check for .uae files with the base name as the adf/zip
        if [[ -f "$rom_path/$rom_bn.uae" ]]; then
            config="$rom_path/$rom_bn.uae"
        elif [[ -f "conf/$rom_bn.uae" ]]; then
            config="conf/$rom_bn.uae"
        else
            if [[ -z "$config" || "$config" == "adf" ]]; then
                if [[ "$name" =~ AGA|CD32 ]]; then
                    config="conf/rp-a1200.uae"
                else
                    config="conf/rp-a500.uae"
                fi
            elif [[ "$config" == "auto" ]]; then            
                config_param = "--model "
                if [[ "$name" =~ AGA|CD32 ]]; then
                    config="A1200"
                else
                    config="A500"
                fi
            fi
        fi
    else
        # add conf directory
        config="conf/$config"
    fi
```

midwan, 2021-05-29:
Update: fixed a bug with RPI3 (possibly also RPI2/1 but not RPI4!) showing a black screen after the GUI, with the latest version.
The fix is deployed in both `master` and `dev` branches.

joolswills, 2021-10-01:
Thanks. Will look into it and do a PR.

midwan, 2022-04-02:
It's been a while since this was opened, and in the meantime Amiberry v5.0 was released yesterday. :)

I've also added a wiki section mentioning how one could manually upgrade to it, in RetroPie (since I've been getting a lot of requests about that): https://github.com/midwan/amiberry/wiki/Upgrading-to-the-latest-version

Any ETA on when this could be integrated in RetroPie-Setup?

VincentGR76, 2022-04-02:
Any news about the latest installation script?
I still have 3.3v on my retropie.

joolswills, 2022-04-02:
Will look into this next. I have been very busy.

nemo93, 2022-04-03:
@midwan thanks for the excellent follow-up and all the work done on Amiberry. Still amazing to see how much have been added/fixed! I suggest to change the ticket title for Preparation for Amiberry v5.x ;-)

@joolswills: happy to help. I've booked the coming days to work on that upgrade. I should submit a PR soon (if you don't beat me to it) where we could work from there. My only concern is about the 2 config files mentioned in the other ticket #2973.

joolswills, 2022-04-03:
Thanks but I'd like to do this so please dont submit a PR.

midwan, 2022-04-17:
Today, Amiberry v5.1 was released (https://github.com/midwan/amiberry/releases/tag/v5.1). 

Please use this tag (or higher) for RetroPie, as it includes several fixes for issues discovered in the Dispmanx version (which RetroPie uses) after V5.0

nemo93, 2022-05-04:
PR has now been merged in successfully. Thanks to everyone involved as this was a big step. Do you mind me suggesting to close that ticket off as well as #2973?

closed 2022-05-04 by midwan